KiCad/kicad-doc #3: po4a-translate cannot process cvpcb.adoc

Fat-Zer, 2015-01-03:
Po4a-translate hangs with an error on cvpcb:
```
po4a-translate -f asciidoc -M utf-8 -m cvpcb.adoc -p po/fr.po -k 0 -l cvpcb-fr.adoc
po4a::asciidoc: Unable to parse attribute list: []
```

```po4a --version
po4a version 0.45.
written by Martin Quinson and Denis Barbier.

Copyright (C) 2002, 2003, 2004 Software in the Public Interest, Inc.
This is free software; see source code for copying
conditions. There is NO warranty; not even for
MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

Works ok on kicad.adoc.
I suspect it's some sort of a po4a bug because asciidoc processes the file well...

Fat-Zer, 2015-01-03:
Hmm.. you already fixed it in cbec1053f7b593750b8e1fcc7ad8c9bcc1fde13e..
I was just played around with old version... 

The current variant works only because of combination of bugs in the po4a asciidoc module:
On the one hand it doesn't handle empty argument list for macros and on the other it ignores in-line macros at all...

Fat-Zer, 2015-01-03:
But there is the same problem with eeschema.adoc

Fat-Zer, 2015-01-03:
Filled in a bug with a fix for upstream:
https://alioth.debian.org/tracker/index.php?func=detail&aid=314919&group_id=30267&atid=410622

ciampix, 2015-01-05:
Many thanks for this patch, I am impressed!

closed 2015-04-06 by ciampix