KiCad/kicad-doc #51: CMake Build doesn't build include files

BrianSidebotham, 2015-05-21:
At the moment, the CMake build system is broken because it does not deal with asciidoc includes which must be generated for each manual chapter and language.

BrianSidebotham, 2015-05-21:
Working on this in https://github.com/BrianSidebotham/kicad-doc

BrianSidebotham, 2015-05-21:
In order to modify the files to correct the include file names it's intended to add a custom target that will run a cmake script to modify the translated file in the binary path. CMake isn't great for this type of work directly in a custom target, so the script is necessary to do this work.

Should work fine though...

nickoe, 2015-05-21:
(I have still not found the time to look through how it really works yet)  But don't a2x normally handle the includes by itself ?

ciampix, 2015-05-21:
On Thu, May 21, 2015 at 02:25:04PM -0700, nickoe wrote:
> (I have still not found the time to look through how it really works yet)  But don't a2x normally handle the includes by itself ?

Simple answer: a2x yes, po4a not (I filed a bug but I doubt that it will be taken into account...)

-- 

Marco Ciampa

I know a joke about UDP, but you might not get it.

+------------------------+
| GNU/Linux User  #78271 |
| FSFE fellow       #364 |
+------------------------+

BrianSidebotham, 2015-05-22:
Nick, Although a2x deals with includes, we need translated versions of the includes and therefore the files get suffixed with the two letter language. Because of that, the includes fail. We need to transform the include filename to the correct name for each language.

nickoe, 2015-05-22:
Ahh ok, I understand the issue now.

BrianSidebotham, 2015-05-25:
Just to note: I did decided against modifying the include filename in the source file - which seems a bit awkward (At least, for CMake this is near impossible!). So instead, I re-organised the build so that the original include filenames work okay.

Now CMake generates language directories full of that language's documentation. This seems more succinct anyway.

TODO: CMake can have install components. We should create a component per language so that tarballs can be made for each language easily. Not everyone needs documentation in all languages!

nickoe, 2015-05-25:
Yeah, I am just building to test it. You might open another issue for that todo item, if you don't intend to fix it immediately.    But how should that work?  Then `make install-pl`?

BrianSidebotham, 2015-05-25:
Thanks Nick,

Re: TODO: It's something like that yes, otherwise we'll have to do a configure-based language selection. I'll open an issue anyway so that we can keep track of progress and ideas.

nickoe, 2015-05-25:
I am not exactly sure why, but when I do: `mkdir build; cd build; cmake ..; make DESTDIR=/tmp/foobar/ install;`

I get a dot in the path as: `-- Installing: /tmp/foobar/usr/local/./en/Pl_Editor/pdf/Pl_Editor.pdf`

closed 2015-05-25 by nickoe

BrianSidebotham, 2015-05-25:
Probably because the install paths are relative to the root install folder, so I always start with `./` which is fine because `/tmp/foobar/usr/local/./en/Pl_Editor/pdf/Pl_Editor.pdf` is exactly equivalent to `/tmp/foobar/usr/local/en/Pl_Editor/pdf/Pl_Editor.pdf`

So it can be safely ignored.

PR #57 (2015-05-25): Fix Issue #51 (CMake bulid incomplete)
This fixes #51 by re-organising the build so that documents are created in a language-specific folder rather than have a language-specific filename. This allows asciidoc includes to be used correctly without fear of having to modify the asciidoc document in order to "force" them to work.

This solution is much better suited to a CMake build system rather than trying to modify the include files. It does not impact on the original Makefile method.

This solution is a little wasteful in including a copy of the images folder in all language html install folde [... cut]
commit 00ff0dd5 (2015-05-25): * Fixes #51 by re-organising build