KiCad/kicad-doc #63: CMake package version is always 1.0.0

BrianSidebotham, 2015-05-27:
The CMake build system package target `make package` always uses the same version number - 1.0.0

1. How should the numbering system go?
2. Should it have a version at all?
3. Will it be aligned with the KiCad "stable" product?

ciampix, 2015-05-27:
IMHO I think that it should be aligned with the KiCad version it refers to (like GIMP manual for instance).

BrianSidebotham, 2015-05-28:
I also think keeping aligned is a good idea. I wonder if we should mark the version as the git branch. That way, a build from the master branch will get you `kicad-doc-master.tar.gz`, etc. and then when a stable release is almost ready with a known triplet, we can create a new branch such as `4.0.0` or whatever. That branch will end up with a package `kicad-doc-4.0.0.tar.gz`, etc.

This is probably the best way of keeping aligned and is also a good way of being able to work on a stable version's documentation.

The other option is to use tags. I'm sure there's a way of git getting the tag, coupled with the number of commits since the tag so we could end up with normal builds being the last stable tag, plus a number indicating the number of commits since that tag, such as `kicad-doc-4.0.0-15.tar.gz`, etc. That may also be good. I prefer branches to tags, but it's really the preference of the translators that matters most here I guess.

nickoe, 2015-05-28:
Instantly I am more inclined to use tags, and you should be able to look up the tag from some commit.  But when I think about it, I guess it is better to use branches, because the docs could possibly lag the release, and then updates specifc to that version could added just to that branch and alternatively also applied on the master.

mangelajo, 2015-05-29:
Thinking of autobuilds, it's easier to work with branches.

We could build testing to http://docs.kicad-pcb.org/testing/....
and stables to http://docs.kicad-pcb.org/stable-xx.xx.xx/.......

Please think of autobuilding and having everything available online in an automated fashion, no human interaction but pushing to branches... :)

I'm all for tags if that can be archived also with tags, then I don't care ;)

danger89, 2015-05-29:
I agree with Brian. The Github tagging is the most logical approach in this case. It makes more sense to create versions, then create multiple branches.
@nickoe  Branches are used to create separate copies and develop in a seperate branch, later you can merge it to the master branch. In this case you always want to have a history, so please choose for tags.

The tags/versions of the document corresponds to a version of KiCad.

danger89, 2015-05-29:
Maybe take a look at Linux as an example:
https://github.com/torvalds/linux

"Better well stolen than badly invented." They say here.

nickoe, 2015-05-29:
The problem with tagging only, is that we only have the linked list of commits (a master branch), but if there is developed a new feature in KiCad that the master docs describe those should not be available in an updated version (backport) of the docs for the lastest stable KiCad. This will happen if we just use plain tagging on master. But I guess tags on a kicad aligned version branch could be used.

This would also mean, that we should tag the docs as, <kicad-version>-<some-doc-append-version>.  Like doc version 3.0.0-b or what ever indication.  I think just replacing the latest tarball of docs could be problematic if packagers would just download the precompiled docs.

nickoe, 2015-05-30:
But the answer the original first three questions: 1. Still to be exactly defined. 2. Yes! 3. I think it should follow the KiCad triplet (assuming that is the numbering system). But we should have some way to indicate backports on the docs. I don't expect all documentation to be ready before the KiCad release, and I expect that we wan't to backport information, even though the main KiCad source has the "no backporting" strategy.

Regarding the technical implementation of getting the version information; I have been thinking a bit more, and I have come to the conclusion that tags are better. They don't care if we branch or not. and it will make only one entry point for cmake to fetch the version info. We can still branch with tags. Tags are bound to commits.

mangelajo, 2015-05-30:
Branches are useful to backport doc fixes across branches. You can combine it with tagging.

Otherwise explain me, if you keep different versions of docs (because software diverges and features diverge), how're you going to backport doc upgrades or fixes on general features over different versions?.

Or you don't plan on keeping stable/testing docs separate?, that'd make not much sense IMHO.

nickoe, 2015-05-30:
Yes of course we need to branch when a stable release is released. But if we have more fixes for an "old" release doc, we shall not need to branch again, just create a new tag a couple of commits later.

So we need the cmake script to be able to generate a folder named the version and have the docs in there.

BrianSidebotham, 2015-06-01:
I think the answer as far as CMake is concerned so far is this:

If git cannot be found, set the package version to -unknown

If `git describe --tags` returns something, set the package version to the result ( This is the last tag found within a branch )

Otherwise, get the current branch name `git rev-parse --abbrev-ref HEAD` and set the package version to the result.

The only thing that's not covered is intermediate versions, i.e. between tagged versions...

nickoe, 2015-07-06:
Yes, I think this is a good idea. Please go ahead.

BrianSidebotham, 2015-07-07:
@nickoe - Will do

BrianSidebotham, 2015-08-12:
I'm going to fix this issue in the next couple of days - sorry for the delay, other projects came to the fore...

commit 4d7142b2 (2015-09-12): Merge pull request #146 from BrianSidebotham/master

Use git info for package version. Fixes #63
commit b0636882 (2015-09-12): Use git info package version. Fixes #63
closed 2015-09-12 by nickoe