KiCad/kicad-doc #73: cmake build always rebuilds all

cpavlina, 2015-06-09:
When building using a cmake-generated Makefile, make rebuilds the whole kit and caboodle, even if only one file was changed since the last make. This is quite slow on some systems!

nickoe, 2015-06-09:
Mentioning @BrianSidebotham in case he is not notified and he wants to help fix this issue.

BrianSidebotham, 2015-06-11:
Thanks @nickoe I had missed this.

cpavlina, I agree - the dependency list is not complete. This is something to be completed, it's not that straight forward with CMake as far as I remember because of the custom commands, but it certainly should be doable.

The things I can recommend for you are:

1. make defaults to the `all` target and so will clearly rebuild all. If you're only editing one document, you can simply just make that target instead of all. e.g. `make CvPcb`, etc.
2. Configure with [SINGLE_LANGUAGE](https://github.com/ciampix/kicad-doc/blob/master/cmake-readme.adoc#single_language) flag

onitake, 2015-09-28:
There's another implication to this: If you use `make install` to copy the built documentation into a staging directory, it will also rebuild everything from scratch. That can be slightly annoying.

BrianSidebotham, 2015-09-29:
In order to fix this problem we're going to have to move from `add_custom_target()` to `add_custom_command()`

The change will be a bit intrusive and will probably require a bit of testing before it's right so I suggest we deal with this problem after a stable release. Things are working okay at the moment even if they are a little annoying.

This should be fixed though, it's not a quick build process to build all the docs!

nickoe, 2015-09-29:
Thatis where theSI

onitake, 2015-09-29:
Thanks @BrianSidebotham, these are exactly my thoughts on the topic.