KiCad/kicad-doc #75: Documentation includes static list of contributors, not updated

cpavlina, 2015-06-09:
All the documentation contains something like this:

    *Contributors*
    
    Jean-Pierre Charras, Fabrizio Tappero.

Quite a few more people have contributed documentation, but nobody remembers  to add names to the list. Perhaps we can either remove the explicit lists in favor of just referring to "the KiCad documentation contributors", or find a way to have that pulled out of the git history automatically?

nickoe, 2015-06-09:
Yeah, I have been thinking about that before, but I have done nothing about it because I want to hear other peoples opinion. I don't quite have one yet.

BrianSidebotham, 2015-06-11:
I think we'll want to end up with something like `git log --stat --pretty=format:'%cN'` (or similar, there's a million possibilities here) and then pass it to some CMake foo which can generate a list of contributors, even on a per-document AND per-language basis.

Caerbannog, 2015-06-28:
Do we need the contributor list in the first place? The git repository is public, we could just link it.

nickoe, 2015-11-28:
Did we ever reach a consensus or decision about this?

heitorPB, 2018-11-03:
What about using the output of `git shortlog --summary --numbered` to make a contributors page instead of having the name of only a few people in each page?

ciampix, 2018-11-03:
Oh nice command! I can't say... you see, I'm biased ... ;-)

heitorPB, 2019-06-13:
Just notice that my name disappeared from the git history as https://github.com/KiCad/kicad-doc/pull/680 was squash-merged. Maybe this happened some other times as well? Another problem to get the contributors from git is that not every user adds their real name here.

Maybe a `contributors.txt` with everybody alphabetically is better?

ciampix, 2019-06-14:
So just a `git shortlog --summary |gawk '{print $2}'|sort`  should suffice?

heitorPB, 2019-07-22:
What about `KiCad Documentation Team` with a link to https://github.com/KiCad/kicad-doc/graphs/contributors ?

heitorPB, 2019-07-23:
I just noticed that in KiCad -> Help -> About there's a list of documentation writers. There are 6 names in there. There's also a list of devs, artists, translators and packagers. Should we find out a way to keep in sync the list in the software and also in the docs?

heitorPB, 2019-07-26:
I just found out the project [AllContributors](https://allcontributors.org/) to help in this task.