KiCad/kicad-doc #83: po4a-updatepo removes numeric lists

Caerbannog, 2015-06-20:
It seems that po4a-updatepo currently removes the numeric lists from the documentation.

Compare these two generated files :
http://ci.kicad-pcb.org/job/any-kicad-doc-head/lastSuccessfulBuild/artifact/src/Getting_Started_in_KiCad/Getting_Started_in_KiCad-fr.html#draw-electronic-schematics
http://ci.kicad-pcb.org/job/any-kicad-doc-head/lastSuccessfulBuild/artifact/src/Getting_Started_in_KiCad/Getting_Started_in_KiCad.html#draw-electronic-schematics

The faulty command is at line 93 of the Makefile and amounts to this:
   po4a-updatepo -f asciidoc -v -M utf-8 -m Getting_Started_in_KiCad.adoc -p po/fr.po

For the following source:
1. Under Windows run kicad.exe. Under Linux type 'kicad' in your ...

The output becomes:
"Under Windows run kicad.exe. Under Linux type 'kicad' in your ...
https://github.com/ciampix/kicad-doc/blob/master/src/Getting_Started_in_KiCad/po/fr.po#L462

I haven't found a fix yet, does someone have a better understanding of the issue?

ciampix, 2015-06-21:
Sorry it is a known bug (or a feaure?) of po4a that works well only with well formatted (i.e. indented) numbered lists.

nickoe, 2015-06-21:
Maybe it should be reported to po4a to get fixed in the future?

ciampix, 2015-06-21:
Yes I should have already reported it in a mail but I forgot when and I am not able to find it now... I'll look for bug reports before filing one...

Caerbannog, 2015-06-21:
My last comments before closing this issue:

@ciampix Your fix left some "  +" before images, they should probably be removed.

Remaining warnings:
- I haven't figured why we still have numbering problems:
```
asciidoc: WARNING: Getting_Started_in_KiCad.adoc: line 363: list item index: expected 1 got 25
asciidoc: WARNING: Getting_Started_in_KiCad.adoc: line 368: list item index: expected 1 got 25
asciidoc: WARNING: Getting_Started_in_KiCad-it.adoc: line 368: list item index: expected 1 got 25
asciidoc: WARNING: Getting_Started_in_KiCad-pl.adoc: line 363: list item index: expected 1 got 25
```

- Missing lang-pl.conf, I know you and @keruseykaryu submitted a merge request to asciidoc, but we might want to commit it in our repo in the meantime
- Issue #84

ciampix, 2015-06-22:
1) the + should only be as the first char of a line to create manual wrap-around so they wanted and it is not a bug.

2) those about list items are warning, not errors

3) I asked the asciidoc for the missing merge of lang-pl.conf. They responded that they do not have any polish dev to confirm the request :-(

4) po4a handle without problems the header underlined chars, using the correct length, so the problem reported simply does not exist. One could argue that prepending "n*=" chars before the headers titles as an alternative syntax is easier and less error prone, and I may agree but this is not vital IMHO.

nickoe, 2015-06-22:
> 1) the + should only be as the first char of a line to create manual
> wrap-around so they wanted and it is not a bug.
>
> 2) those about list items are warning, not errors
>
> 3) I asked the asciidoc for [...]
Maybe ask them to verify with google translate then.

Caerbannog, 2015-06-22:
1. Sorry I was talking about an intermediary commit.
2. Allright I understand those warnings. The following paragraph:
    ```
20. Hover the mouse over the microcontroller component. Press the y key
    or the x key on the keyboard. Notice how the component is flipped over
    its x axis or its y axis. Press the key again to return it to its
    original orientation.
```

    Gets transformed after Polish translation into :
    ```
20. Najedź myszą na mikrokontroler. Naciśnij klawisz *Y* lub klawisz *X* na
klawiaturze. Zauważ w jaki sposób komponent zostanie przerzucany w osi X lub
Y. Naciskaj klawisze ponownie by wrócić do jego oryginalnego położenia.
```

    And after Italian translation into :
    ```
20. Portare il puntatore del mouse sopra il componente microcontrollore. Premere
il tasto 'y'. Si noti come il componente viene ribaltato sul suo asse x o
y. Premere nuovamente il tasto per riportarlo al suo orientamento originale.
```

    We're very unlucky, both have a line starting with "Y. ", which is the 25th letter of the alphabet. Asciidoc thinks we're starting a new list, hence the "list item index: expected 1 got 25".

    The result is here (look for "Premere nuovamente il tasto") :
    http://ci.kicad-pcb.org/job/any-kicad-doc-head/lastSuccessfulBuild/artifact/src/Getting_Started_in_KiCad/Getting_Started_in_KiCad-it.html#draw-electronic-schematics

3. OK.
4. I insist, look at the first items in the table of contents:
    http://ci.kicad-pcb.org/job/any-kicad-doc-head/lastSuccessfulBuild/artifact/src/Pl_Editor/Pl_Editor-pl.html
    
    And compare it to:
    http://ci.kicad-pcb.org/job/any-kicad-doc-head/lastSuccessfulBuild/artifact/src/Pl_Editor/Pl_Editor.html

keruseykaryu, 2015-06-22:
@ciampix: 3) I asked the asciidoc for the missing merge of lang-pl.conf. They responded that they do not have any polish dev to confirm the request :-(
@nickoe: Maybe ask them to verify with google translate then.

Let them come out into the street, will certainly find some there. :-)

@Caerbannog: Polish translation might be simple refine to:

    20. Najedź myszą na mikrokontroler. Naciśnij klawisz *Y* lub klawisz *X* na
    klawiaturze. Zauważ w jaki sposób komponent zostanie przerzucany w osi X lub
    osi Y. Naciskaj klawisze ponownie by wrócić do jego oryginalnego położenia.

Repetition of the word "osi" (_EN: axis_) isn't that much of a grammatical issue.

kinichiro, 2015-09-24:
This warning still remains.
<pre>
asciidoc: WARNING: getting_started_in_kicad.adoc: line 362: list item index: expected 1 got 25
</pre>
And I noticed that just removing the word ` orientation` from line 362 solves this warning.

```diff
diff --git a/src/getting_started_in_kicad/getting_started_in_kicad.adoc b/src/getting_started_in_kicad/getting_started_in_kicad.adoc
index 40c8afc..f8dee89 100644
--- a/src/getting_started_in_kicad/getting_started_in_kicad.adoc
+++ b/src/getting_started_in_kicad/getting_started_in_kicad.adoc
@@ -359,7 +359,7 @@ be connected. We will see later on why this is the case.
 20. Hover the mouse over the microcontroller component. Press the y key
     or the x key on the keyboard. Notice how the component is flipped over
     its x axis or its y axis. Press the key again to return it to its
-    original orientation.
+    original.

 21. Repeat the add-component steps, this time choosing the 'device'
     library and picking the 'LED' component from it.
```
Is there any special meanings for the word "orientation" in asciidoc ?

LordBlick, 2015-09-24:
Try to binary compare… Some windows extra char or sth else…

ciampix, 2015-11-24:
Is there still this problem? May I close this?

ciampix, 2016-06-13:
No answer in 6 months ... closing.

closed 2016-06-13 by ciampix