KiCad/kicad-doc #84: Translation ruins title because length changes

Caerbannog, 2015-06-20:
The doc currently uses the following format for titles:
    Introduction to *Pl_Editor*
    -------------------------

Here is the correponding title in Pl_Editor-pl.adoc:
    Wprowadzenie do programu *Pl_Editor*
    -------------------------

According to asciidoc's documentation: "underlines consist a repeated character pairs spanning the width of the preceding title (give or take up to two characters)".
The result is a mangled structure, compare those:
http://ci.kicad-pcb.org/job/any-kicad-doc-head/lastSuccessfulBuild/artifact/src/Pl_Editor/Pl_Editor-pl.html
http://ci.kicad-pcb.org/job/any-kicad-doc-head/lastSuccessfulBuild/artifact/src/Pl_Editor/Pl_Editor.html

I propose to convert all docs to "One line titles" of the form:
    === Section title (level 2) ===

nickoe, 2015-07-06:
Is that the only way to fix it?   And that two line title is with dashes, and should be a level 1 title as far as I can see.   Do you have any other suggestions?

I agree that constraint with the specific number of indicator chars under the title is not a good option for translatable docs, so it might indeed be better with one line titles.  I would like to hear @ciampix 's experince on this matter.

Caerbannog, 2015-07-06:
I don't see any other solution, the reference is here: http://www.methods.co.nz/asciidoc/userguide.html#X17

It seems the end tag is optional so both of the following are valid, although number 1 is probably clearer:
```
=== Section title 1 ===
=== Section title 2
```

We might want to convert all titles before languages proliferate.
Or just this instance which is indeed a level 1.

nickoe, 2015-07-06:
@Caerbannog so you say that it is only the level 1 that suffers from that constraing or ?

Caerbannog, 2015-07-06:
No, sorry. I meant two things:
- Confirm your sentence: "And that two line title is with dashes, and should be a level 1 title as far as I can see."
- The problem manifested itself only once so far (`Wprowadzenie do programu Pl_Editor`), you can check for "section title out of sequence" in [the warnings](http://ci.kicad-pcb.org/job/any-kicad-doc-head/lastSuccessfulBuild/warnings2Result/). So we might want to fix only this perticular title (`Introduction to Pl_Editor`).

nickoe, 2015-07-06:
If you make a pull request on that I will merge this. But lets wait with doing it for the rest untill the vast ammount of japaneese translations is merged and @ciampix comes back.

kinichiro, 2015-07-13:
+1

japanese translation also has same issue, like this.
http://ci.kicad-pcb.org/job/kicad-doc-testing-pull-request/99/artifact/build-cmake/src/Pl_Editor/en/Pl_Editor.html
http://ci.kicad-pcb.org/job/kicad-doc-testing-pull-request/99/artifact/build-cmake/src/Pl_Editor/ja/Pl_Editor.html

replace should be like below ?
<pre>
 2 line titles  ->    1 line titles
Level 0: ==== ->  = Document Title (level 0) =
Level 1: ---- ->  == Section title (level 1) ==
Level 2: ~~~~ ->  === Section title (level 2) ===
Level 3: ^^^^ ->  ==== Section title (level 3) ====
Level 4: ++++ ->  ===== Section title (level 4) =====
</pre>

kinichiro, 2015-07-15:
I wrote the perl script for converting from two-line-title to one-line-title.
https://gist.github.com/kinichiro/b07f113b991951a1c4a2

This script is not perfect, so before running this script, 
you should shorten to half the '------...' line below, by hand.
 - Eeschema_creating_customized_netlists_and_bom_files.adoc 372
 - IDF_Exporter.adoc 388

original .adoc is renamed to .adoc.org.

<pre>
# conv_two2one.pl CvPcb/*.adoc
# conv_two2one.pl Eeschema/*.adoc
# conv_two2one.pl GerbView/*.adoc
# conv_two2one.pl Getting_Started_in_KiCad/*.adoc
# conv_two2one.pl IDF_Exporter/*.adoc
# conv_two2one.pl KiCad/*.adoc
# conv_two2one.pl Pcbnew/*.adoc
# conv_two2one.pl Pl_Editor/*.adoc
</pre>

nickoe, 2015-07-20:
Ahh yes, there it was.  But I don't understand what you mean with shorten to half. (I will have a look at converting them all later today [hopefully])

If you want to, I will be fine with it if you can convert all two line headings to one lines.

Caerbannog, 2015-07-20:
I didn't see @kinichiro's script before writing this one:
https://gist.github.com/Caerbannog/9714617f3879ddcc741d
It will be interesting to see if some corner cases are handled differently.

After reading [asciidoc's recommended practices](http://asciidoctor.org/docs/asciidoc-recommended-practices/#section-titles) I now think we should use this shorter version of titles, except maybe for level 0:
```
== Title 1
```

kinichiro, 2015-07-20:
Hi,
I tried to replac two-line-title to one-line-title at this my local branch.
https://github.com/kinichiro/kicad-doc/tree/mod_section
I used "shorter version of titles" method.

And you could see the differences before/after by this.
https://github.com/ciampix/kicad-doc/compare/master...kinichiro:mod_section?diff=split&name=mod_section

I tested by creating pdf, and I checked by myself.
Take a look at, please.

Thanks.

Caerbannog, 2015-07-20:
I tested it and i'm OK.

kinichiro, 2015-07-20:
Thanks for your testing. I PRed https://github.com/ciampix/kicad-doc/pull/120 .

nickoe, 2015-07-20:
Fixed in #120

kinichiro, 2015-07-21:
I will update po and PR them.
Could you merge PR https://github.com/ciampix/kicad-doc/pull/117 before that, please ?
It already passed the test.

nickoe, 2015-07-21:
Sure.

kinichiro, 2015-07-21:
I PRed for updating all po files by PR https://github.com/ciampix/kicad-doc/pull/121 .

nickoe, 2015-07-21:
Merging #120 and #121 closes this issue.

closed 2015-07-21 by nickoe