KiCad/kicad-doc #92: Move to kicad/kicad-doc

Caerbannog, 2015-06-21:
Should new pushes go to kicad/kicad-doc?
(it seems to lag behind at the moment)

Should the issues be transferred too?
It seems possible: https://help.github.com/articles/transferring-a-repository/

nickoe, 2015-06-21:
I don't know, maybe it is @stambaughw trying to move it over. Maybe we will see something on the list shortly.

stambaughw, 2015-06-22:
It wasn't me.  I see someone already transferred (or maybe they just
cloned which wasn't the correct way to do this) the kicad-doc repo to
github:\\KiCad.

kinichiro, 2015-07-03:
Is this a formal upstream for kicad-doc activities or not ?
Kicad/kicad-doc seems stop since Jun-25th.
Which repo to align our translation tasks ?

nickoe, 2015-07-03:
For now use this one (ciampix).

ciampix, 2015-08-27:
Repo transfered.

closed 2015-08-27 by ciampix