KiCad/kicad-doc #99: can't open xcf in Pl_Editor with latest gimp 2.8

kinichiro, 2015-07-02:
Hi,

Now I'm trying to translate 
and I can't open 2 xcf files within Pl_Editor.
I had got, xcf version 8 is unsupported error.
I'm using gimp 2.8, latest.

How did you edit xcf within PI_Editor ?

Thanks

ciampix, 2015-07-02:
On Wed, Jul 01, 2015 at 08:38:43PM -0700, kinichiro inoguchi wrote:
> Hi,
> 
> Now I'm trying to translate 
> and I can't open 2 xcf files within Pl_Editor.
> I had got, xcf version 8 is unsupported error.
> I'm using gimp 2.8, latest.
> 
> How did you edit xcf within PI_Editor ?
> 
> Thanks
> 

Sorry sorry sorry!

I develop (translate) also for GIMP and I inadvertently opened it with
GIMP dev version (2.9). I'll try to reimport it into GIMP 2.8 xcf, please
hold on, or I'll communicate if it is impossible...

Please forgive my adsent mind!

-- 

Marco Ciampa

I know a joke about UDP, but you might not get it.

+------------------------+
| GNU/Linux User  #78271 |
| FSFE fellow       #364 |
+------------------------+

kinichiro, 2015-07-02:
Hi, @ciampix 

I see. No problem.
I will do another translation task,
And I'm waiting you.

Thanks for your work for everything.

Best regards.

kinichiro

ciampix, 2015-07-02:
On Thu, Jul 02, 2015 at 12:51:45AM -0700, kinichiro inoguchi wrote:
> Hi, @ciampix 
> 
> I see. No problem.
> I will do another translation task,
> And I'm waiting you.
> 
> Thanks for your work for everything.
> 
> Best regards.
> 
> kinichiro

Done. git pull and try use the template now...

-- 

Marco Ciampa

I know a joke about UDP, but you might not get it.

+------------------------+
| GNU/Linux User  #78271 |
| FSFE fellow       #364 |
+------------------------+

kinichiro, 2015-07-02:
Hi, your 2 new xcf works fine.
Thanks.

closed 2015-07-02 by kinichiro