KiCad/kicad-doc #104: Some parts of Eeschema .adoc appears NOT to be copied to .po file

kinichiro, 2015-07-08:
Some parts of Eeschema .adoc appears NOT to be copied to .po file.
I checked it.po, ja.po and pl.po, but couldn't find the 2 parts below.

(1) Eeschema_generic_commands.adoc ; section 2.3

        76 * The “?” key displays the current hotkey list.
        77 * Hotkeys can be managed by choosing "Edit Hotkeys" in the Preferences menu.
        78
        79 Here is the default hot key list:
        80
        81 image::images/en/default_hot_key_list.png[Default hotkey list]
        82
        83 All hot keys can be redefined by the user via the hotkey editor:
        84
        85 image::images/en/shortcuts_editor.png[Hotkeys editor window]

(2) Eeschema_design_verification_with_erc.ado ; section 8.4 

        74 It is common to have an error or a warning on power pins, even though
        75 all seems normal. See example above. This happens because, in most
        76 designs, the power is provided by connectors, that are not power sources
        77 (like regulator output, which is declared as Power out).
        78
        79 The ERC thus won't detect any Power out pin to control this wire and
        80 will declare them not driven by a power source.
        81
        82 To avoid this warning you have to place a “PWR_FLAG” on such a power
        83 port. Take a look at the following example.
        84
        85 image:images/20000008000030E4000026DDFDF3D5E2.png[20000008000030E4000026DDFDF3D5E2_png]
        86
        87 The error marker will then disappear.
        88
        89 Most of the time, a PWR_FLAG must be connected to GND, because usually
        90 regulators have outputs declared as power out, but ground pins are never
        91 power out (the normal attribute is power in), so grounds never appear
        92 connected to a power source without a pwr_flag.
        93
        94 [[configuration]]

Is this a known issue or not ?

Best Regards.

kinichiro, 2015-07-20:
I've found this issue is also caused by issue https://github.com/ciampix/kicad-doc/issues/84 ,
so called, two-line-title problem.

**Eeschema_generic_commands.adoc : line 73**
<pre>
...

Hotkeys
~~~~~~~~

* The “?” key displays the current hotkey list.
* Hotkeys can be managed by choosing "Edit Hotkeys" in the Preferences menu.
...
</pre>
There are 8 '~' , but 'Hotkeys' is 7 chars.

**Eeschema_design_verification_with_erc.adoc : line 71**
<pre>
...
[[power-pins-and-power-flags]]
Power pins and Power flags
~~~~~~~~~~~~~~~~~~~~~~~~~~~

It is common to have an error or a warning on power pins, even though
...
</pre>
There are 27 '~' , but strings above have 26 chars.

nickoe, 2015-07-20:
Ok, thank you for figuring this out. Did someone make a script to convert the title format?

kinichiro, 2015-07-20:
If no one did it, I wrote perl script. see issue https://github.com/ciampix/kicad-doc/issues/84#issuecomment-121625243 .

kinichiro, 2015-07-21:
PR https://github.com/ciampix/kicad-doc/pull/120 had resolved this issue, too.

closed 2015-07-21 by kinichiro