KiCad/kicad-doc #105: diffs in descriptions about max layer number

kinichiro, 2015-07-08:
Hi,

description about max layer number in KiCad.adoc seems wrong.

- KiCad.adoc
<pre>
84 Kicad does not present any board-size limitation and it can
85 handle up to 32 copper layers, 14 technical layers and 4 auxiliary layers.
</pre>

- Getting_Started_in_KiCad.adoc
<pre>
85 KiCad does not present any board-size limitation and it can
86 easily handle up to 16 copper layers and up to 12 technical layers.
</pre>

- Pcbnew_introduction.adoc
<pre>
61 Pcbnew works on 16 layers of copper, plus 12 technical layers (silk
</pre>

- Pcbnew_layers.adoc
<pre>
5 Pcbnew can work on 29 different layers:
6
7 * 16 layers of copper (or of routing of tracks)
8 * 12 auxiliary technical layers.
9 * 1 board outlines layer
</pre>

Thanks.

nickoe, 2015-07-08:
KiCad.adoc is correct.  You can count it yourself.   Recently support for up to 32 copper layers was added and some new technical layers was added.

kinichiro, 2015-07-08:
Then, GSiK and Pcbnew .adoc are needed to be fixed ?
They are saying up to 16 copper layers, now.

nickoe, 2015-07-08:
Ohh that is what you mean with gsik.

Yes, those should be updated, please make a pull request.

kinichiro, 2015-07-09:
PR was merged and I will close this issue.
Thanks.

closed 2015-07-09 by kinichiro