KiCad/kicad-doc #106: Japanese version error in table

nickoe, 2015-07-08:
There is a formatting issue in the table in section 2.8. Upper toolbar, it seems that the line
https://github.com/ciampix/kicad-doc/blob/44cc91a0882fc264e92b8a32763b324aaa3243d4/src/Eeschema/po/ja.po#L2774

is containing some spurious characters, `a|\n"` which should be removed. @yoneken Can you please fix this (such that we make sure that I don't mess up the alignment). Also I see multiple strings in that file named "EeSchema" while it should just be "Eeschema" to be in accordance with the rest.

yoneken, 2015-07-08:
Of course, I can remove it.
However, "a|¥n" is come from the original version.

Should I remove the characters only japanese version?
4a1c55930772c852fcf33d3cd68762175262ec5c Is it OK?

nickoe, 2015-07-13:
The EeSchema to Eeschema fix should apply everywhere.  Also I would like you to fix that table, such that I don't screw up on that one.

kinichiro, 2015-07-21:
merged PR https://github.com/ciampix/kicad-doc/pull/117 had resolved this issue, too.
see table in section 2.8 of pdf below.
http://ci.kicad-pcb.org/job/kicad-doc-testing-pull-request/109/artifact/build-cmake/src/Eeschema/ja/Eeschema.pdf

closed 2015-07-21 by nickoe