KiCad/kicad-doc #107: Build is broken

BrianSidebotham, 2015-07-08:
I think the Japanese merge ( #100 ) was too quick, now the build fails for me with various errors to do with xelatex and fonts. Please update the cmake-readme.adoc documentation to detail any added dependencies and also please fix the build.

The add_dblatex_option() works globally, so now all languages are being built with the xetex backend and with the japanese style sheet.

The japanese style xsl should go in the already-existing /xsl folder or else move (and re-reference) the kicad.xsl (used by FOP) file from the /xsl folder into the new src/style folder.

As a side, to support building with FOP we may need more style sheets, but we can probably deal with that at a later date.

nickoe, 2015-07-08:
Hmm. The japaneese style is already in the xsl folder. Maybe I pulled the trigger on the japanese merge too quickly. I just verified that both pdf's had working fonts.

Are you looking into fixing this issue or is the work open for anyone else to grab? (Me, myself, I go to bed now)
Regarding FOP, what style sheets are we talking about here? More xsl files?

BrianSidebotham, 2015-07-08:
Anyone's welcome! I just noted it before I had to get to bed! Lol

I thought the style sheet had gone under the /src/style folder rather than
/xsl.

We could, for now, make the Japanese support something the is configured as
an option, I.e. -DJAPANESE=ON or something which would solve the problem
while we fix the issues. Having individual dblatex options (i.e.
stylesheets, etc) for individual languages will be the one that takes the
most effort to do, but is probably worth it in the long run anyway.

The Japanese translation is an excellent effort, I'm sure we'll get the
build back to normal quickly.

yoneken, 2015-07-09:
Sorry for the confusion.
I took care my change will not to work globally (e.g. https://github.com/ciampix/kicad-doc/blob/master/CMakeModules/KiCadDocumentation.cmake#L85-L88).
It's not enough?

nickoe, 2015-07-09:
Hm, yes indeed, I think that is enough.   When I open 

http://docs.kicad-pcb.org/Eeschema/Eeschema-ja.pdf
vs
http://docs.kicad-pcb.org/Eeschema/Eeschema.pdf

you will see that the japanese contains the vlgothic fonts and lmroman for the rest, while the english use the nimbus font.  I guess that is fine then?

You can also try to compare the Eeschema word on the top of the pages.

BrianSidebotham, 2015-07-09:
@nickoe `Eeschema-ja.pdf` is created by the Makefile build system, not the CMake build system, you'd be looking for `Eeschema/ja/Eeschema.pdf` from a CMake build to do a comparison.

It's easy to test by enabling the `--verbose` a2x option in the root `CMakeLists.txt` which will then show all of the command lines. Here's some output from a failed build with that option enabled and `make VERBOSE=ON > log.txt 2>&1` on my Mint Linux box (Split into an option per line to see it more easily):

    a2x: executing: "dblatex" -t pdf
    -p "/etc/asciidoc/dblatex/asciidoc-dblatex.xsl"
    -s "/etc/asciidoc/dblatex/asciidoc-dblatex.sty"
    -P latex.output.revhistory=0
    -P doc.publisher.show=0
    -s /home/brian/kicad/kicad-doc/ciampix/kicad-doc/src/pdf-cover-dblatex.sty
    -b xetex
    -p /home/brian/kicad/kicad-doc/ciampix/kicad-doc/src/CvPcb/../style/dblatex-pdf-ja.xsl
    -o /home/brian/kicad/kicad-doc/ciampix/kicad-doc/build/src/CvPcb/en/CvPcb.pdf
    -V  "/home/brian/kicad/kicad-doc/ciampix/kicad-doc/build/src/CvPcb/en/CvPcb.xml"

So, the `xelatex` backend is being used for all languages. As I don't yet have the correct fonts to support Japanese it all falls over with messages like: 

    kpathsea: Running mktextfm VL-Gothic-Regular/ICU
    /usr/share/texlive/texmf-dist/web2c/mktexnam: Could not map source abbreviation I for ICU.
    /usr/share/texlive/texmf-dist/web2c/mktexnam: Need to update /usr/share/texlive/texmf-dist/fonts/map/fontname/special.map?
    mktextfm: Running mf-nowin -progname=mf \mode:=ljfour; mag:=1; nonstopmode; input ICU
    This is METAFONT, Version 2.718281 (TeX Live 2013/Debian)

    kpathsea: Running mktexmf ICU

    ! I can't find file `ICU'.
    <*> \mode:=ljfour; mag:=1; nonstopmode; input ICU
                                                    
    Please type another input file name
    ! Emergency stop.
    <*> \mode:=ljfour; mag:=1; nonstopmode; input ICU
                                                    
    Transcript written on mfput.log.
    grep: ICU.log: No such file or directory
    mktextfm: `mf-nowin -progname=mf \mode:=ljfour; mag:=1; nonstopmode; input ICU' failed to make ICU.tfm.
    kpathsea: Appending font creation commands to missfont.log.

Unfortunately `add_dblatex_option()` works on a global options list, not a per-language or per-target options list. We will need to extend the functionality to achieve that.

nickoe, 2015-07-09:
Ahh yes, that is correct. I guess that the build is not copying the files properly yet to the artifacts.

But looking at: http://ci.kicad-pcb.org/job/any-kicad-doc-head/ws/build-cmake/src/Eeschema/en/Eeschema.pdf

It looks ok to me, what is the issue here?

I now see that we have xsl files in two places, `src/style/dblatex-pdf-ja.xsl` and `xsl/kicad.xsl`

Would it be bad to make cmake force you to get CJK fonts? (at least by default)

Is it possible to detect such things properly with cmake?

BrianSidebotham, 2015-07-09:
The issues here are detailed above - nothing to do with the ci server.

CMake should detect the required dependencies and continue if they're available or else barf with an error about the missing dependency. Building all documents with the ja style is also not right. We've just killed everyone's CMake build unless they have whatever CJK fonts are required.

The FOP build is also broken, but only is so far as the ja documentation is filled with `#` instead of any Japanese characters.

yoneken, 2015-07-12:
I think it's not a critical issue.

You can avoid the build error at least 3 ways.

1. Install a japanese font (VLGothic)
  Easiest way. I suggest this way.
2. Avoid  to build japanese documents
  There is a function to select which language will you build  in CMakeLists.txt (SINGLE_LAUNGUAGE variable).
3. Change the build process to load cmake build by build

BrianSidebotham, 2015-07-13:
Well this is critical for me because it breaks my build. With reference to your points:

1. Please add that to the documentation then (cmake-readme.adoc) so everyone knows the vlgothic font is a pre-requisite of building the documentation. It was introduced by #100 but isn't documented. FOP doesn't require this in order to build, it simply builds on Windows replacing the missing glyphs with # - for now we have to suffer the build not creating the right output on Windows, but at least the build isn't broke.

2. That doesn't mean I can successfully build because that only builds one language. I know about the option because I wrote the option.

3. Sorry - I don't know what you mean by that.

Lastly, we shouldn't be applying a `-ja` suffixed style sheet to all documents. (Can be solved simply by renaming the style sheet and putting the `add_dblatex_option()` into the main `CMakeLists.txt` file so it's obvious it's being applied to all languages.

nickoe, 2015-07-13:
Is the font only required for the latex build? If so I guess the requirement should be listed under the dblatex section.

I am sorry that I broke the system with merging #100.

If you have the overview to fix the above issues, that would be nice... I do know time is a limited resource.

PR #116 (2015-07-15): Fix #107
Dear @BrianSidebotham , @nickoe ,
I added the required information about building japanese pdfs to fix #107 .

I would appriciate if you have a broad-mindedness for minority languages.
commit 09e19a35 (2015-07-15): Merge pull request #116 from kicad-jp/fix_107

Fix #107
closed 2015-07-15 by nickoe