KiCad/kicad-doc #115: module to footprint

kinichiro, 2015-07-13:
the word `module` might be `footprint` in the new KiCad manuals.
I tried to think of the table to transform.

obsolete                              | new                 |
--------------------------------------|---------------------|
ModEdit, Module Editor, module editor | Footprint Editor    |
module libraries                      | footprint libraries |
.mod                                  | .lib                |
module                                | footprint           |

I've just greped the .adoc files, 
and I found these 4 documents will be the target.
- Getting_Started_in_KiCad
- KiCad
- IDF_Exporter
- Pcbnew

I put this issue, but this seems very hard to work on this.

nickoe, 2015-07-13:
I guess we should ay least do this soon. But while doing that one might to review the information in those sections, because they might easily be outdated.

But .mod is not .lib.   The lib file is for the schematics.  The med footprint format is kicad_mod. So it should probably just be replaced in 99% of the cases, except from.where we might talk about how to use the legacy footprints.

kinichiro, 2015-07-15:
If you do this, could you tell us about the schedule, please ?
Changing the .adoc file leads re-translating.
So, if we know what week .adoc will change,
we are able to have a plan for translating.

Say,
Jul-15 - 17 : .adoc change and commit or merged.
Jul-18 - 22 : translation period, .adoc never change.

I feel .adoc and .po should sync in short span.
For this purpose, .adoc period and .po period 
are required, IMHO.

Thanks

nickoe, 2015-07-21:
@kinichiro Regarding the issue about fixing in the English docs breaks the translated docs: I think we can't not really hold back on the English on the master. I think when we have a release we can call that the translation period, because then the main docs should settle a bit more. I am open to different opinions.

kinichiro, 2015-07-21:
@nickoe
for non native English reader, translated document is the document.
so, all translators would like to deliver the well translated one as far as they could.
"well" means "describe the things right", here.
good document makes someone to be interested more and more.
sometimes, it has more value than the software itself. (kidding)
then, I would like to have a time to "translate the things right" after changing English document.

on the other hand, yes, I understand it is difficult to hold back on the English on the master.
this is the global open source project, involves many person around the world, and project has a due date.

conclusion, I wish I could have a time to fix translation between last English doc changes and project release time ;-)

Thanks for your hard work for everything on this project.

ciampix, 2015-07-27:
My personal and very humble opinion is that the translation is rarely a one time effort but... GIMP docs stayed immutable for more that a year so translations are really in good shape and complete in 10 (!) different languages.

The thing is: we have a shortage of document authors so the translators are actually doing the author's work but not in a systematic / omni-comprehensive way. This is a bad thing especially since docs was never so difficult to write so the consequences are that now there is an urgent need to revise docs. 

I think we have to ask or at least exhort devs and volunteers to contribute to the docs in order to bring the docs at least to an updated state. After that period we can freeze docs (actually a real "string freeze" as is used to do with programs before official versions come out) or using git to create a branch or a tag and ask for translators to complete at least the Getting Started in KiCad docs in a reasonable period of time. Then we can always have on-line the "updated" translations of the "official" (i.e. freezed) docs and a running version of the docs that follows the dev trunk of KiCad program.

So I agree with @kinichiro but we do not have to "hold back" the English on the master but we have to create a "snapshot" window to allow translators to work without having to update every day for a reasonable period of time.

But I repeat: this is only my very humble opinion, having paucity of doc writers, devs, native language revisers and translators/revisers every strategy have some pros and cons...

nickoe, 2015-07-27:
What about this?

Whe have the master doc, the english doc. We then have that rolling.

Then the translaters should be able to say which version is most recent, this can be done with a tag. This means that we have freezed the latest version of a given translated doc untill the translator get the translations upleveled.  This essentially means that the translators are responsible of tagging a snapshot which does not have po files that does not match the english in that snapshot.

I think this can work well, but we will need to device a practical way of doing this. If people think this is a good idea, I will try to document this principle and how it is going to work in detail. This including versioning to match the kicad releases.

kinichiro, 2015-07-28:
+1
Thanks @ciampix for your "snapshot" idea.
I would like to propose one thing that the live snapshot should be only one at a time.

nickoe, 2015-07-28:
@kinichiro  does that +1 mean that you agree with my suggestion, which is a more complete description of a solution, or do you disagree with it in any way?

kinichiro, 2015-07-28:
@nickoe I do agree with your suggestion. 
And I would like to hear other contributors opinion, too.

Caerbannog, 2015-07-28:
Just to make sure everyone agrees that ensuring that translated docs are published without residual English strings does not mean that any .po or .adoc should be frozen in `master`.
A solution could be to give translators the option of creating branches dedicated to polishing.

The build system could have a switch to build only "release-ready docs" from these branches. Or maybe this is just a matter of configuring Jenkins, and pushing the right thing to docs.kicad.org.

nickoe, 2015-07-28:
@Caerbannog That is intended to be taken care of in my suggestion. But I think a tag should be enough. I don't like unnessesary branching, although I need to think it through to see if we really need it in some cases or not.

kinichiro, 2015-10-17:
Original issue (module -> footprint) was fixed by PR above.
Another topics about managing and versioning of docs were written in 
https://github.com/KiCad/kicad-doc/blob/master/docs-versioning.adoc by @nickoe.
Can we close this issue ?

closed 2015-10-17 by nickoe