KiCad/kicad-doc #122: unification of :doctype:

kinichiro, 2015-07-21:
`:doctype:` is different between each manuals.

- `book` : Eeschema, GerbView, Getting_Started_in_KiCad, Pcbnew
- `ariticle` : CvPcb, IDF_Exporter, KiCad and Pl_Editor

I feel that the `book` is suitable for the manuals.

GUI_Translation_HOWTO.adoc seems not a `book`, 
so this should be an `article`, I feel.

Caerbannog, 2015-07-25:
What changes do you want?

From what I understand the "book" just creates in the PDF output dedicated "chapters" for each level 1 title. My opinion:
- `Eeschema`, `Pcbnew` : Those should still be books, they look great as PDFs.
- `Getting_Started_in_KiCad` : I have no opinion, so keep it as a book.
- `KiCad` : Might be converted to a book.
- `CvPcb` : Might be converted to a book, but since it's not an separate app anymore I wouldn't bother.
- `IDF_Exporter`, `GUI_Translation_HOWTO`, `Pl_Editor` : Article works fine here.
- `GerbView` : I submitted PR #130 to convert it to an article because it consists of a single chapter since we removed the Gerber specs.

kinichiro, 2015-07-25:
`:doctype:` affects both PDF and EPUB for displaying a chapter.
I thought that all the manuals should be "book" when I posted this issue,
since I prefer to style of "book" for formal documentation :smile: 

But after seeing your categorize in webdocindex (Main and Additional),
I could understand your way, too.
And then, `CvPcb` should keep "article".

Conclusion of my thoughts
- `KiCad` : Might be converted to a book.
- `GerbView` : Might be converted to a article. already PR #130

Thanks.

closed 2015-09-14 by ciampix