KiCad/kicad-doc #123: lists with line break and po4a

kinichiro, 2015-07-23:
lists with line break and po4a has some trouble.
for example, CvPcb.adoc,
<pre>
     87 * Cvpcb needs the project config file to know the footprint libraries to load.
     88 * Cvpcb initializes the components footprint fields of the current schematic project.
     89 This is possible only when Eeschema is running.
</pre>
this asciidoc generates correct "list of 2 items" output **without po4a**, like this.
- Cvpcb needs the project config file to know the footprint libraries to load.
- Cvpcb initializes the components footprint fields of the current schematic project. ...

but **with po4a**, it generates "list of 1 item", like this.
- Cvpcb needs the project config file to know the footprint libraries to load.
  Cvpcb initializes the components footprint fields of the current schematic project. ...

to get correct output, 2nd line of item with line break should be indented, like this.
(see line 89)
<pre>
     87 * Cvpcb needs the project config file to know the footprint libraries to load.
     88 * Cvpcb initializes the components footprint fields of the current schematic project.
     89   This is possible only when Eeschema is running.
</pre>

I wrote perl script to find this kind of lists in .adoc files.
https://gist.github.com/kinichiro/ec1e7dea649c600e5dd5
run `./invalid_lists.pl kicad-doc/src/*/*.adoc` , then you could see how many lists like this issue.

nickoe, 2015-07-23:
Hmm, interesting.  I think this is related to https://github.com/ciampix/kicad-doc/issues/83

Maybe we should apply the above mentioned workaround and document it in the README or something.

kinichiro, 2015-07-23:
yes, I also think this relates to issue 83.
shall I submit PR for this ?

nickoe, 2015-07-23:
Yes, that would be fine. I just ran the script, but I did not start fixing it.

kinichiro, 2015-07-24:
I've found another pattern of list issue.
case of 2 level list, in Pcbnew section 2.4.7, 3.1, 3.2.1, 3.4, 5.2.1, 5.2.3, 10.3.3 and 14.3.2.2.

example is here.
po4a seems can't handle conjunction of _*_ and _**_.
<pre>
     11 * left toolbar menu.
     12 * mouse buttons (menu options). Specifically:
     13 ** The right-hand mouse button reveals a Pop up menu the content of
     14 which depends on the element under the mouse arrow.
     15 * keyboard (Function keys F1, F2, F3, F4, Shift, Delete, +, - Page Up,
     16 Page Down and “space” bar). The “Escape” key generally cancels an
     17 operation in progress.
</pre>

workaround : inserting line break for each list item could resolve this issue.
<pre>
     14 * left toolbar menu.
     15
     16 * mouse buttons (menu options). Specifically:
     17
     18 ** The right-hand mouse button reveals a Pop up menu the content of
     19    which depends on the element under the mouse arrow.
     20
     21 * keyboard (Function keys F1, F2, F3, F4, Shift, Delete, +, - Page Up,
     22   Page Down and “space” bar). The “Escape” key generally cancels an
     23   operation in progress.
     24
</pre>

kinichiro, 2015-07-25:
Thanks for merging PR, and I will close this.

closed 2015-07-25 by kinichiro