KiCad/kicad-doc #124: "NOTE" paragraph

kinichiro, 2015-07-23:
for paragraph "NOTE", you should use `NOTE:` or `[NOTE]`.
but in .adoc files, we could see these examples below,
- `*Note:*`
- `*Note*`
- `=== Note`
- `==== Notes`
- `Notes:`
- `    NOTES:`
maybe, these are mistakes.

kinichiro, 2015-07-24:
I picked up some questionable use of "Note" in the adoc files.
But almost of them might not be harmful,except Eeschema section 6.3.2 title.

adoc file name						|expression	|section|
--------------------------------------------------------|---------------|-------|
CvPcb/CvPcb_footprint_libraries_tables.adoc		|\*Note:*	|5.2	|
Eeschema/Eeschema_schematic_creation_and_editing.adoc	|\*Note:*	|5.5.3	|
Eeschema/Eeschema_schematic_creation_and_editing.adoc	|Note:		|5.5.4	|
Eeschema/Eeschema_hierarchical_schematics.adoc		|==== Notes	|6.3.2	|
Eeschema/Eeschema_hierarchical_schematics.adoc		|\*Note*	|6.7	|
Eeschema/Eeschema_design_verification_with_erc.adoc	|Notes:		|8.2	|
Eeschema/Eeschema_create_a_netlist.adoc			|\*Note*	|9.2	|
Eeschema/Eeschema_create_a_netlist.adoc			|=== Note	|9.4	|
Eeschema/Eeschema_component_library_editor.adoc		|\*Note:*	|11.4	|
Eeschema/Eeschema_component_library_editor.adoc		|\*Note:*	|11.4.1	|
Eeschema/Eeschema_component_library_editor.adoc		|\*Note:*	|11.4.1	|
Eeschema/Eeschema_component_library_editor.adoc		|\*Note:*	|11.4.1	|
IDF_Exporter/IDF_Exporter.adoc				|\*Note:* In the current revision of |4.4 |
Pcbnew/Pcbnew_zones.adoc				|==== Notes	|10.9.2	|

Caerbannog, 2015-07-24:
Converting all of those to the dedicated syntax would certainly be an improvement.

kinichiro, 2015-07-24:
@Caerbannog thanks, I will submit this as PR with some other improvements.

ciampix, 2015-08-27:
All suggestion applied, thanks, so I close this issue.

closed 2015-08-27 by ciampix