KiCad/kicad-doc #129: The "Footprint Libraries Wizard" in pcbnew needs to be mentioned in the libraries seciton

nickoe, 2015-07-25:
The "Footprint Libraries Wizard" in pcbnew needs to be mentioned in the libraries seciton

kinichiro, 2015-09-03:
Does this issue mean which of below ?

- section 5.3 "Footprint Libraries Wizard" of CvPcb should be placed in chapter 12 of Pcbnew.

or

- whole chapter 5 "Footprint Libraries Management" of CvPcb (CvPcb_footprint_libraries_tables.adoc) should be transferred to Pcbnew.

nickoe, 2015-09-03:
It is just a matter of telling the reader of the footprint libraries wizards existence when one read about footprint libraries. I don't think sections needs to be moved.

kinichiro, 2015-09-03:
Oh, I see.
Then, would you like to add the section like below at the end of Pcbnew_managing_libs.adoc (chapter 12) ?

<pre>
=== Footprint Libraries Management

Pcbnew uses a library management tool based on *_footprint library tables_* .
You can manage many kinds of footprint library, and add new one to footprint library tables
by *_Footprint Library Table Wizard_* .

More details about footprint library tables and Footprint Library Table Wizard,
see _CvPcb Reference manual "5. Footprint Library tables"_ .
</pre>

nickoe, 2015-09-03:
Sort of like that, but I think you are mixing the two concepts too much in the first sentence.

cpavlina, 2015-09-03:
=== Footprint Libraries Management
    
    The list of footprint libraries in Pcbnew can be edited using the Footprint Libraries Manager.
    This allows you to add and remove footprint libraries by hand, and also allows you to invoke
    the Footprint Libraries Wizard by pressing the "Append With Wizard" button.

    The Footprint Libraries Wizard can also be invoked through the Preferences menu, and can
    automatically add a library (detecting its type) from a file or from a Github URL. The URL for
    the official libraries is: _https://github.com/KiCad_
    
    More details about footprint library tables and the Manager and Wizard can be found in the
    CvPcb Reference Manual: "5. Footprint Library Tables".

Someone will have to linkify the cvpcb reference as I don't remember how and don't have a chance to check right now.

kinichiro, 2015-09-03:
Indeed, my draft mentions about table and wizard.

cpavlina, 2015-09-04:
"Table" is an internal term, most of the UI doesn't use it and it'll confuse people. The language also reads a bit awkwardly, it's not idiomatic for a user manual.

kinichiro, 2015-09-04:
Sorry, I refer to CvPcb section 5, then I used the word Table.
http://docs.kicad-pcb.org/CvPcb/CvPcb.html#_footprint_libraries_management

My draft was not good :-)

@nickoe How do you like draft of @cpavlina ?

BTW, cross ref link across the different docs (pcbnew and cvpcb) is available ?

And, are there any functions should be described in the reference docs before stable is released ?
This might create as another issue.
 
Thanks

ciampix, 2015-09-04:
Well, assuming that the doc main address will not change for a while (http://docs.kicad-pcb.org) and the doc name will not change also (like, for instance, CvPcb), inserted Internet absolute links will work as cross reference between docs. This could have big drawbacks though since it jumps from a local copy to an Internet and possibly updated version. The best way to handle this should be to have relative links but this would not work (if it works at all) with different format of documents like pdf and epub. So I really do not know which is the best way™ to handle this problem...

nickoe, 2015-09-04:
@kinichiro, i like @cpavlina.  I am away for the weekend and I expect @ciampix to merge it.

ciampix, 2015-09-05:
Ok I copypasted and committed the @cpavlina version. If it's ok I will close this.

closed 2015-09-08 by nickoe

nmz787, 2015-10-13:
the link posted by kinichiro doesn't work (http://docs.kicad-pcb.org/CvPcb/CvPcb.html#_footprint_libraries_management)

nickoe, 2015-10-13:
Use http://docs.kicad-pcb.org/en/cvpcb.html#_footprint_libraries_management