KiCad/kicad-doc #131: Improve documentation forewords

Caerbannog, 2015-07-25:
Since the forewords is the first thing that people see, we might want to polish it.
It's mostly similar between documents, see [GSIK](http://ci.kicad-pcb.org/job/any-kicad-doc-head/ws/build-cmake/src/Getting_Started_in_KiCad/en/Getting_Started_in_KiCad.html) and [KiCad](http://ci.kicad-pcb.org/job/any-kicad-doc-head/ws/build-cmake/src/KiCad/en/KiCad.html).

Suggestions:

1. The document title is duplicated in bold. Remove it:
   ```
GUI Translation HOWTO
=====================

*GUI Translation HOWTO* => remove
```
2. Issue #75 discusses the "Contributors" lists. I propose to replace it with a link to the commit history. Or to have CMake generate a CONTRIBUTORS.txt because it's easier.
3. The "Feedback" part should point to Github issues.
4. Remove "Acknowledgments" because it's systematically empty.
5. Publication date and software version is a good idea, but it's out of date. Suggestions of what CMake should generate?
6. Should the "Note for Mac users: ... support is ... experimental." be removed now?

I will try to move quickly on these points when I have some feedback.

nickoe, 2015-07-25:
I will try to reply to the items:
1. I think it would be ok to remove this.
2. There has been no conclusion to this topic. I think linking to authors would be ok. But I would really like and opinion from the original authors. Jean-Pierre Charras, Fabrizio Tappero.
3. Agree, although it might also be usefull to make it point to the bug tracker on lp, for software issues.
4. Agreed.
5. I think this should be replaced with what is generated from a solution to https://github.com/ciampix/kicad-doc/issues/63.
6. I don't care.

kinichiro, 2015-07-25:
Let me add one.
7. in copyright statement, use single byte `(C)` and `-`. now © and – are used.

nickoe, 2015-07-25:
Ad 7. I do not rember seeing the dash. But I think we should make it such that it renders correctly the symbol in the end docs, although I don't know where it is used.

kinichiro, 2015-07-26:
Ad.7  multi byte dash is used between year and year in copyright statement.
asciidoc should be a plain ascii text, I think.

nickoe, 2015-09-12:
@Caerbannog What is the status on this?

ciampix, 2015-09-12:
1. was added to produce the effect that I see in many printed books, of the title re-printed in the first page, and even in HTML it is useful, as I see, to have the title repeated on the first page other that in the browser title bar. So I would leave it as it is.
2. no comment but I mostly agree with @nickoe 
3. agree with @nickoe 
4. agree
5. agree with @nickoe
6. agree
7. agree with @kinichiro 

4 and 6 I would do it right now...

ciampix, 2015-09-12:
Sorry I was seeing the Getting started in Kicad, the only doc _without_ the title repeated. Ok to delete it. I'll do it right now.

ciampix, 2015-09-13:
Committed point 1, 4, 6 and 7 on all docs.

Caerbannog, 2015-09-13:
Perfect. The last important point is the publication date / version. Commit b0636882 recently created a `CPACK_PACKAGE_VERSION` string. This obviously can't be used with the legacy `Makefile` system, so is any transition planned for `docs.kicad-pcb.com` ?

ciampix, 2015-09-14:
The same string format could be created also in the Makefile "way", I do not see any real problem there. I still do not understand how we can manage to include the date / version in the asciidoc source file without some sort of search & replace.

nickoe, 2015-09-14:
But we should move away from the manual makefiles and only use cmake.

ciampix, 2015-09-14:
Yes I understand and agree but, my fault as usual, I'm a bit frustrated because I am of no use in it because I am (still!!) not able to contribute in cmake so I tend to continue to support what is working and what I am able to do.

Anyway I am not flattened on using Makefiles instead of cmake but I have some problems here (Linux Ubuntu) that I am not able to handle probably due to my ignorance again. I have some errors if I use the cmake sequence following the README (cd built; cmake ../; make), errors that I do not see doing a make all-all using my Makefile:

[...]
Built target GerbView_html_ja
a2x: ERROR: "dblatex" -t pdf -p "/etc/asciidoc/dblatex/asciidoc-dblatex.xsl" -s "/etc/asciidoc/dblatex/asciidoc-dblatex.sty" -P latex.output.revhistory=0 -P doc.publisher.show=0 -s /home/marco/git/github/kicad-kicad-doc/src/pdf-cover-dblatex.sty -b xetex -p /home/marco/git/github/kicad-kicad-doc/src/GerbView/../../xsl/dblatex-pdf-ja.xsl -o /home/marco/git/github/kicad-kicad-doc/build/src/GerbView/ja/GerbView.pdf  "/home/marco/git/github/kicad-kicad-doc/build/src/GerbView/ja/GerbView.xml" returned non-zero exit status 1
make[2]: *** [src/GerbView/CMakeFiles/GerbView_pdf_ja] Errore 1
make[1]: *** [src/GerbView/CMakeFiles/GerbView_pdf_ja.dir/all] Errore 2
[...]

I should probably open a bug on this or am I doing something really wrong here (as usual)?

ciampix, 2015-09-14:
... no comment (please) :-(. Found a forgiven bash alias in my home stettings on make (make -j4) to speedup compilation but dblatex is _not_ multithread safe... sorry for the noise...

ciampix, 2015-09-14:
Well, BTW, why I find that in (for example) build/src/CvPcb there are all languages dirs /en /fr /it etc. and in _every_ dir there are all languages files (CvPcb_footprint_libraries_tables-fr.adoc, CvPcb_footprint_libraries_tables-it.adoc, CvPcb_footprint_libraries_tables-ja.adoc, etc)? Is this happening just in my linux build environment?

ciampix, 2015-09-14:
About point 5: what do we do? Asciidoc have a nice sys call that could be used to get system date and such but I would avoid it for security reasons (and because asciidoctor does not support it) and use a simple include instead. We could manage to prepare a simple asciidoc compatible txt file created during compilation time with the git commit version (with the same format and logic used to create the archive name in the cmake macros) & system date inside it and include it into the source with the include command. That would be compatible with asciidoctor too. What do you think?

nickoe, 2015-09-25:
@Caerbannog FWIW we have archived the old cmake system and should only focus on the cmake build system. The old makefile is kept in the repo for easy reference.

kinichiro, 2015-10-05:
About point 6., "(still experimental, but is now working well)" should be removed in src/kicad/kicad.adoc
<pre>
     79
     80 KiCad is available for Linux, Windows and Apple OS X (still experimental, but is now working well).
     81
</pre>

And point 1., 4., 6. and 7. were fixed by @ciampix ,
then point 2., 3. and 5. still remain.

kinichiro, 2015-10-05:
For point 3. ("Feedback"), how do you like this draft ?
For GSiK, KiCad, Eeschema, Pcbnew, CvPcb, GerbView, Pl_Editor, IDF_Exporter
and gui_translation_how_to.
<pre>
*Feedback*

Please direct any bug reports, suggestions or new versions to here:
- About KiCad document : https://github.com/KiCad/kicad-doc/issues
- About KiCad software : https://bugs.launchpad.net/kicad
- About KiCad software i18n : https://github.com/KiCad/kicad-i18n/issues
</pre>

ciampix, 2015-10-08:
I like it. @nickoe? Any doubts? Drawbacks on rely on github? What do you think?

kinichiro, 2015-11-12:
Shall we solve this issue, since this is the last milestone for 4.0.0 ?

ciampix, 2015-11-12:
If there are not people against this I will apply last @kinichiro suggestion because seems that the other issues are already resolved and close this PR.

Caerbannog, 2015-11-12:
Yes, please close it.

closed 2015-11-12 by ciampix

kinichiro, 2015-11-12:
Thanks @ciampix !

commit 54df2027 (2015-11-12): Updated globally Feedback section of docs, closes #131