KiCad/kicad-doc #132: Improve images and screenshots

Caerbannog, 2015-07-25:
I suppose we could do a better job at keeping images consistent with each other in a single document, and updated with the latest KiCad version.
My proposals:

1. Icons can be exported directly from the KiCad sources. Currently they are either mixed with screenshots under `src/<DOCUMENT>/images/` or in a separate `src/<DOCUMENT>/images/icons`. Proposal: move all of them under `src/icons` to avoid duplication and make sure they keep their original name.

2. The screenshots are usually arbitrary, so updating them while keeping the meaning isn't easy. Proposal: use always the same example project, like the hackRF demo. Or a specific board commited here, with all features. That's lot of work.

3. Screenshots have names like `100000000000010700000125A4376EBB` that probably date from the ODT export. We could give them proper names.

   Maybe try to name them with instructions on how they where made like `eeschema-hackrf-con1-contextmenu.png`.

4. The PDF output sometimes has image DPI inconsistencies that @ciampix addressed in [src/AUTHORS_README.adoc](../blob/master/src/AUTHORS_README.adoc), someone should have a look at it.

5. Minor subject: OS and window manager inconsistencies.

 What are your ideas on these and other subjects?

nickoe, 2015-07-25:
I will reply to each item:

1. I would like to not duplicate files, especially images. And yes, they must keep the original name matching the kicad source.
2. I don't think this is super critical. Can you point us at specific sections where this is an issue. IMHO it does not make sense to waste time, space or other resources making sure the the screenshots are made on every platform. And I prefer if people try to minimize the file size of images. Often images can be reduced quite a bit by reducing the color depth.
3. This is a work in progress, and should preferable be done by those who update the relevant sections. A side effect is that you can search for those image names and find sections that are probably outdated.
4. I am not exactly sure what this means. But I think we should investigate a2x sizing comands for images. I think images wider than the view should _often_ be just scaled to fit the textwidth. Images smaller should be "real" sized.
5. I don't think it matters that there are used different window decorations and widget styles for the screenshots. I quite like that they are different. It shows that kicad runs on many platforms. Although it might not be as aesthetically pleasing for the designer, but we should really focus on the content. But I do think that the screenshot should be cropped to not include the window manager.

Caerbannog, 2015-07-26:
On item 1:
I have tried to move the `icons/` directory directly in `src/` but then HTML generation needs relative paths like `image:../../icons/abc.png`.
This messes with ePub generation (`a2x: ERROR: resource file outside destination directory: ../../icons/abc.png`) and I haven't found a solution. We might need to add `--resource <somedir>` to `a2x` or change CMake's WORKING_DIRECTORY.

nickoe, 2015-07-26:
Alternatively the install target should just copy the files.

kinichiro, 2015-07-29:
about 4. 
as mentioned in [src/AUTHORS_README.adoc](https://github.com/ciampix/kicad-doc/blob/master/src/AUTHORS_README.adoc), screenshot images should be formatted by parameter.
I tried to use block(`image::`) macro with `scaledwidth=NN%`.
It works good almost, but doesn't work in a table.
In a table, inline(`image:`) macro with `width="NN%"` seems to work.

I did this with Pl_Editor by PR https://github.com/ciampix/kicad-doc/pull/138 .
take a look at [Pl_Editor.pdf](http://ci.kicad-pcb.org/job/kicad-doc-testing-pull-request/125/artifact/build-cmake/src/Pl_Editor/en/Pl_Editor.pdf).

If this way of adjusting image is OK, I would like to do the same thing on GerbView, too.

ciampix, 2015-07-30:
Nice catch! I have no problem accepting this, @nickoe ?

PS: some (Pl_Editor page 11 / 19 - 7.2 Main Window Toolbar) table images looks weird though...

nickoe, 2015-07-30:
@ciampix I won't disagree if stuff don't bloat the repo. But regarding those strange tables with images in pl_editor.pdf... I think those are bretter off being reformulated without putting the images in tables. Maybe edit the images to one image in an image editor to get better control of it as a monolitich unit.

ciampix, 2015-07-30:
Sure, I agree 100% with you @nickoe . See... docs need to be reformatted. Many tables are due to bad habits that word-processors used as desktop-publishing induce into users. Converting ODT to asciidoc just revealed these "bugs". I reckon that the poor table support of rest was a conscious decision...

kinichiro, 2015-11-04:
For item 1., it is just an idea, though, how about using symbolic link.

I tried to do below, 
- move all images/\*.png and images/\*/\*.png to under the src/images/.
- create symbolic link under the each doc directories. `ln -s ../images/ .`

Benefits of this method are ...
- not required modification of adoc files.
- cmake build was OK. (I tested on my Ubuntu15.10)

Disadvantages are ...
- preview on github doesn't work. image files are not showed on the browser.
  take a look at my personal branch:
  https://github.com/kinichiro/kicad-doc/blob/move_images/src/eeschema/eeschema_automatic_classification_annotation.adoc
- maybe, doesn't work on Windows since using symbolic link

github appears 
- can manage symbolic link
- can't follow symlinks on web browser

nickoe, 2015-11-04:
Symlinks are not really portable, so I dislike that idea.

kinichiro, 2015-11-04:
Exactly, we should take a consider more portable way.