KiCad/kicad-doc #133: Improve document - Eeschema

kinichiro, 2015-07-26:
add improve for Eeschema by comment.

kinichiro, 2015-07-26:
I checked current software and current PDF.
- [ ] 1.1 DRC -> ERC
- [ ] 4.2 this section (component library editor pref) should be merged to 3.2.4
- [ ] 10.1 File menu image and description are obsolete.
- [ ] 10.3 HPGL icon is obsolete.
- [ ] 10.3 HPGL plot option image is obsolete.
- [ ] 10.3.2 "HPGL Offset adj." not exist.
- [ ] 10.4 PostScript icon is obsolete.
- [ ] 10.4 PostScript plot option image is obsolete.
- [ ] 10.5 SVG plot option image is obsolete.
- [ ] 10.5 SVG icon is obsolete.
- [ ] 10.6 DXF plot option image is obsolete.
- [ ] 10.6 DXF icon is obsolete.
- [ ] 10.? description about PDF format is missing.
- [ ] 11.5.1 Component Library Editor image is obsolete.
- [ ] 11.5.4 Component Library Editor image is obsolete.
- [ ] 12.2 Component Library Editor image is obsolete.
- [ ] 13.2 Library Browser image is obsolete.
- [ ] 13.3 Library Browser tool bar image is obsolete.
- [ ] 14.3.4.1 Add Plugin tab is obsolete

kinichiro, 2015-07-31:
Let me ask about this issue.
Who fix these description issues ?
Original writer ? Project writer ? Everyone here ?
I PRed some, but those were all for formatting issues, not description.
I wonder who should fix description issues in adoc.

nickoe, 2015-07-31:
Regarding all the icons, I think @Caerbannog would fix that, he made a pull request about such thing a few days ago. But I am not sure that his patchset is ready to merge, I think it requires some more work to not have duplicates of the images in the source.

kinichiro, 2015-07-31:
@nickoe thanks, I will wait and see.

nickoe, 2015-09-25:
I would like to catch up on this task again because I think this is important for our next milestone and we should be able to make new screenshots now that we are in the release candidate phase the GUI should more static.

kinichiro, 2015-09-26:
Are there any new Eeschema functions (since bzr4022) not described yet in manual ?
I think those should be described if any.

nickoe, 2015-09-26:
I don't recall any.

LordBlick, 2015-09-26:
Indeed cvpcb from user point of view become a part of eeschema…

kinichiro, 2015-10-03:
I intentionally left the old screenshots if they were same as 4.0RC1.
I think it is good that shows KiCad is running on multi platform.
But, each size of screenshots are different.
Old ones are big, and new ones (that I captured this time) are small.
I thought small size is good for PDF.

kinichiro, 2015-10-13:
About "4.2 this section (component library editor pref) should be merged to 3.2.4".
When I created the list above, I thought section 3.2.4 and 4.2.1 are duplicate.
But I noticed that 3.2.4 tells about contents of Preferences menu, 
and 4.2.1 enumerates the Options of the schematic editor,
they are different point of view.
I would like to leave this without removing the section 4.2.1,
ok ?

kinichiro, 2015-10-17:
I will close this issue.

closed 2015-10-17 by kinichiro