KiCad/kicad-doc #137: Improve document - Pcbnew

kinichiro, 2015-07-28:
add improve for Pcbnew by comment.

kinichiro, 2015-07-28:
- [ ] 2.3 Preferences menu image is old
- [ ] 2.3 footprint library dialog image is old
- [ ] 2.4 Preferences menu image is old
- [ ] 2.4.6 "Tpically" -> "Typically"

- [ ] 3.1 image is old
- [ ] 3.8 "Inch mm combo icon.png" -> "Inch_mm_combo_icon.png" ('_' is missing)
- [ ] 3.9 image is old (Route menu is missing)
- [ ] 3.9.1 image is old (File)
- [ ] 3.9.2 image is old (Edit)
- [ ] 3.9.3 image is old (View)
- [ ] 3.9.4 should explain Route menu.
- [ ] 3.9.4 this section should be removed since 3.9 explains about top menu.
- [ ] 3.9.5 Pcbnew place menu.png -> "Pcbnew_place_menu.png" ('_' is missing)
- [ ] 3.9.6 image is old (Preferences)
- [ ] 3.9.6 "module libraries" -> "footprint libraries"
- [ ] 3.9.7 image is old (Dimensions)
- [ ] 3.9.8 wrong image and should use Pcbnew_tools_menu.png, but this image is old, too.
- [ ] 3.9.10 this section should be removed since 3.9 explains about top menu.
- [ ] 3.10 top toolbar image is old
- [ ] 3.10 table: "module editor (Modedit)" -> "footprint editor"
- [ ] 3.10 table: row of footprint viewer should be added
- [ ] 3.10 table: row of python scripting console should be added
- [ ] 3.14 2nd row (Footprint mode), icon should be images/icons/mode_module.png
- [ ] 3.14.1 images are old
- [ ] 3.14.2 images are old
- [ ] 3.14.3 images are old

- [ ] 4.1 "The following figure illustrate the whole work-flow of Kicad ..." there is no figure after this.
- [ ] 4.4.1 image is old (netlist dialog)
- [ ] 4.4.2 netlist options should be updated
- [ ] 4.4.3 image is old
- [ ] 4.4.3 "Move all Modules" -> "Spread out All Footprints"
- [ ] 4.4.3 "Move new Modules" -> "Spread out Footprints not Already on Board"

- [ ] 5.1.1 description of number of layers is old.
- [ ] 5.1.2 "Preferences" -> "Design Rules"
- [ ] 5.1.2 "from 2 to 16" -> "from 2 to 32"
- [ ] 5.1.2 images are old
- [ ] 5.1.3 image is old
- [ ] 5.3.1 image is old
- [ ] 5.3.2 image is old
- [ ] 5.3.3 images are old
- [ ] 5.4 image is old

- [ ] 6.1.1 Alt-U -> Ctrl-U (switch units)
- [ ] 6.1.4 image is old (netlist dialog)
- [ ] 6.2.3 images are old
- [ ] 6.2.4 image is old
- [ ] 6.2.5 image is old
- [ ] 6.3 images are old

- [ ] 7.4 popup menu images are old.
- [ ] 7.4 "Move all Modules" -> "Spread out All Footprints"
- [ ] 7.4 "Move new Modules" -> "Spread out Footprints not Already on Board"

- [ ] 8.2 Preference menu image is old
- [ ] 8.2 General Settings image is old
- [ ] 8.3.3 This sentence should enumerate, "Blind/buried Vias", "Micro Vias" and "Minimum Allowed Values"
- [ ] 8.7.1 popup menu image is old
- [ ] 8.8.1 image is old
- [ ] 8.8.1 should describe about Track, Via, Grid and Zoom.
- [ ] 8.8.2 popup menu image is old

- [ ] 10.2.3 popup menu image is old
- [ ] 10.4 popup menu image is old

- [ ] 11.4 images are old.
- [ ] 11.4.3 images are old.
- [ ] 11.4.3 "Use Proper Gerber Extensions" is not exist.
- [ ] 11.4.3 "Subtract Mask from Silk" -> "Subtract soldermask from silkscreen"
- [ ] 11.5.1 Dimensions menu image is old
- [ ] 11.6 image is old
- [ ] 11.9 both image and description are old.

- [ ] 12.* ModEdit -> Footprint editor

- [ ] 13.* ModEdit -> Footprint editor

- [ ] 14.2 Move item images are old
- [ ] 14.3.2 array dialog image is old
- [ ] 14.3.3 array dialog image is old

ciampix, 2015-08-26:
Update some strings and updated previous list accordingly...

nickoe, 2015-09-25:
I have marked this for the milestone, I think we should ensure to handle all the points here before we can call the docs updated.

kinichiro, 2015-09-26:
Are there any new Pcbnew functions (since bzr4022) not described yet in manual ?
I think those should be described if any.

These are descibed already.
- Footprint Library Wizard (described in CvPcb manual 5.3)
- Interactive Router (described in Pcbnew manual 9.)
- Python scripting (described in Pcbnew manual 15.)

nickoe, 2015-09-26:
There is the 3D shapes downloader, but that is very similar to the footprint library wizard. It could probably do with a mention of it in the end of the footprint library wizard.

Do we describe the different canvases anywhere? I think what would be good, maybe in the GSIK?

nickoe, 2015-09-26:
I noticed that the python based footprint wizards has not been described that well, but I tried to make a little description for it in this commit, https://github.com/KiCad/kicad-doc/commit/367e670506099321a13346594ba59c25306cfb87.  Feel free to improve it.

kinichiro, 2015-09-27:
- 3D shapes downloader
  yes, it is similar to the footprint library wizard,
  and it is called only from Preferences menu of pcbnew (right ?).
  So, what about mentioning in pcbnew 12.12 "Footprint Libraries Management" section,
  or adding 12.13 "3D Shapes Libraries Downloader" section ?

- different canvases
  I agree with describing about canvases is good.
  IMHO, this should be mentioned in pcbnew 3.9.3 "View menu" section.
  I think GSIK is a first step tutorial, and details of each functions should be written to each manuals.
  This is just my personal opinion :-)

- python based footprint wizards
  Thanks for adding description. I think it is good.
  To enable url link, we should split it to another paragraph.

nickoe, 2015-09-27:
Fell free to describe the two first points of your last comment.  But about the footprint wizard, what do you mean? We do you mean with this?

kinichiro, 2015-09-27:
about "footprint wizard", take a look at http://docs.kicad-pcb.org/en/pcbnew.html#Footprint_Wizards , please.
you can see link to launchpad is not enabled.

nickoe, 2015-09-27:
@kinichiro I think the URL link is fixed https://github.com/KiCad/kicad-doc/commit/64f86324eca98147c1008647a8297c054a171b45

kinichiro, 2015-09-28:
@nickoe yes, URL link for launchpad is alright.
Will you capture the screen shots with 4.0 RC1 ?
Or upcoming 4.0 RC2 ?

nickoe, 2015-09-30:
@kinichiro I guess you are asking me if we should start updating the screenshots.  Yes, I think it should be fine to start now, there should not be a lot of UI changes at this moment in time. Please go ahead an update the screenshots. But if possible, try to make the windows small enough to fit resonable well on the text when rendered to the PDF version. Then they should also become acceptable in epub and html. I suggest to try to reduce the filesize of the images. I know we use png's, and I like that. But some images can sometimes be halfed in size just by halving the colordepth.

Sorry for the delayed response. I should not have kept you hanging there.

ciampix, 2015-10-01:
PNG could be "indexed" that is: graphic program (for instance GIMP) count colour used in the image and uses just those. In this way image size is greatly reduced.

kinichiro, 2015-10-01:
@nickoe @ciampix I see. I will start this according to list above.

kinichiro, 2015-10-03:
Thanks for merging PR.

I intentionally left the old screenshots if they were same as 4.0RC1.
I think it is good that shows KiCad is running on multi platform.
But, each size of screenshots are different.
Old ones are big, and new ones (that I captured this time) are small.
I thought small size is good for PDF.

About rest of the list above, could anyone write them to adoc, please ?
Sorry, but I'm not good at writing :-)

PR #179 (2015-10-09): improve document - Pcbnew #137 (2)
fixed #137 issuses below.
- 4.1 "The following figure illustrate the whole work-flow of Kicad ..." there is no figure after this.
- 4.4.2 netlist options should be updated
- 8.3.3 This sentence should enumerate, "Blind/buried Vias", "Micro Vias" and "Minimum Allowed Values"
- 8.8.1 should describe about Track, Via, Grid and Zoom.
- 11.4.3 "Use Proper Gerber Extensions" is not exist.
- 11.4.3 "Subtract Mask from Silk" -> "Subtract soldermask from silkscreen"
- 12.* ModEdit -> Footprint Editor
- 12.5 this section should be removed since same as 13.7
- 12.12 lin [... cut]
closed 2015-10-09 by ciampix