KiCad/kicad-doc #152: CvPcb via standaloe Eeschema

kinichiro, 2015-09-09:
This shold be described in CvPcb manual.
https://lists.launchpad.net/kicad-developers/msg20221.html

ciampix, 2015-09-10:
I agree. Probably both in the CvPcb manual and in Eeschema. The CvPcb intro is now definitely wrong: http://docs.kicad-pcb.org/CvPcb/CvPcb.html

##########

It can be run only from Eeschema, form the main toolbar, when started as a project from the KiCad project manager.

the reasons are:

* Cvpcb needs the project config file to know the footprint libraries to load.

* Cvpcb initializes the components footprint fields of the current schematic project. This is possible only when Eeschema is running.

###########

Proposal (RFC: please check if what I say is right):

###########

It can be run either by Eeschema, from the main toolbar, when started as a project from the KiCad project manager, or as a stand alone application.

Running from Eeschema is generally better because:

* Cvpcb needs the project config file to know the footprint libraries to load.

* Cvpcb initializes the components footprint fields of the current schematic project. This is possible only if the project file is in the same path as the open schematic.

Lauching CvPcb from Eeschema assures all this.

You can launch CvPcb as a stand alone application though but please note that any schematic opened that does not have a project file in the same path may be missing components due to missing libraries which will not show up in CvPcb.  If there is no fp-lib-table file in the same path as the open schematic, no project specific footprint libraries will be available either.

###########

This is probably too specific to be told in the CvPcb intro...

nickoe, 2015-09-10:
As far as I can see it does not exist as a stand alone application. The
difference that I see is only that you do not need to start eeschema from
the kicad app to make the cvpcb button appear. Before you needed to start
eeschema from the kicad app, otheriwise the cvpcb button was not available.

ciampix, 2015-09-10:
You are right! Ok let's reformulate it a bit:
[...]
It can be run only from Eeschema, from the main toolbar, either when Eeschema is started from the KiCad project manager or when Eeschema is started as a stand alone application.

Running CvPcb from Eeschema lauched from the KiCad Manager is generally better because:

* Cvpcb needs the project config file to know the footprint libraries to load.

* Cvpcb initializes the components footprint fields of the current schematic project. This is possible only if the project file is in the same path as the open schematic.

Lauching CvPcb from an Eeschema launched from the KiCad manager assures automatically all this.

You actually *can* launch CvPcb from a stand alone Eeschema session though, but please note that any schematic opened that does not have a project file in the same path may be missing components due to missing libraries which will not show up in CvPcb. If there is no fp-lib-table file in the same path as the open schematic, no project specific footprint libraries will be available either. You are have been warned.

nickoe, 2015-09-10:
@ciampix  I think that using the term "main toolbar" is ambiguous, I think "top toolbar" might be better. (We can't move the toolbars as of now.

I think the formulation of the first part of the sentence of the last paragraph is strange. Does it make sense to keep the last statement "You have been warned.", I think that should just be removed.

kinichiro, 2015-09-11:
How about using [warning] paragraph for the whole last block instead of
removing that sentence ?

ciampix, 2015-09-11:
@nickoe 
1) top toolbar, as always you are right and it is consistent with all the docs ok!
2) the suggestion of @kinichiro I think is the best way to handle it.

Let's try it again:

It can be run only from Eeschema, from the top toolbar, either when Eeschema is started from the KiCad project manager or when Eeschema is started as a stand alone application.

Running CvPcb from Eeschema lauched from the KiCad Manager is generally better because:

* Cvpcb needs the project config file to know the footprint libraries to load.

* Cvpcb initializes the components footprint fields of the current schematic project. This is possible only if the project file is in the same path as the open schematic.

Lauching CvPcb from an Eeschema launched from the KiCad manager assures automatically all this.

[WARNING]: You actually *can* launch CvPcb from a stand alone Eeschema session though, but please note that any schematic opened that does not have a project file in the same path may be missing components due to missing libraries which will not show up in CvPcb. If there is no fp-lib-table file in the same path as the open schematic, no project specific footprint libraries will be available either.

PS: please note that comments here in markdown does not support the richer asciidoc syntax so the [warning] paragraph is not rendered correctly.

nickoe, 2015-09-11:
Yes, I think this is much better than the initial formulation.

ciampix, 2015-09-14:
Do you think it is good enough now? May I proceed and commit it or do you think this could be better reformulated?

nickoe, 2015-09-14:
Please commit.

commit 3afafe74 (2015-09-14): Added comment on using CvPcb from a standalone run of Eeschema, fixes #152
closed 2015-09-14 by ciampix