KiCad/kicad-doc #154: CMake version doesn't include master hash and -dirty flag

BrianSidebotham, 2015-09-14:
Two issues:

- The CMake package version doesn't include a Git short hash so the package could be any commit of the docs
- The CMake package version doesn't mark the package as dirty when local uncommitted changes are present

As far as I'm aware we can easily add a git short hash to the package name when Git is present. I propose that we only do this for master and not for the other versions which are built out of branch tags. However, it's opinion, so if more opinion would rather add a Git short hash to all package versions (except `unknown`) then that's fine with me.

A `-dirty` flag is usually pretty easy to know that local changes are present. In BZR this is reasonably easy, although the revno becomes out-of-sync if there are local committed changes. I guess something similar happens here with Git when local changes have been committed so any suggestions on `-dirty` flag generation are welcome. Otherwise I'll come up with something reasonable.

Caerbannog, 2015-09-25:
Regarding `-dirty`, the following command will output `<hash>-dirty` :
```git describe --dirty --always --match foo```

BrianSidebotham, 2015-09-29:
Thanks for the heads up. I'll try and get this sorted out in the next few days...