KiCad/kicad-doc #155: SINGLE_LANGUAGE is not very flexible

BrianSidebotham, 2015-09-14:
The **SINGLE_LANGUAGE** option is not very flexible. It would be better to have an option that is a list of languages which defaults to all languages.

So, I propose we remove the **SINGLE_LANGUAGE** option and instead have a **LANGUAGES** option. To most people building a single language this will just mean reconfiguring with `-DLANGUAGE=en` instead of `SINGLE_LANGUAGE=en` but it does mean that any number of languages could be built, or excluded from a build.

I see this as a pretty low-priority task at the moment, but it would be good to tidy the option up.

nickoe, 2015-09-28:
Yes, indeed, I think a LANGUAGES option is more flexible.

ciampix, 2015-09-28:
Agree

nickoe, 2018-05-05:
Fixed by pr #567

closed 2018-05-05 by nickoe