KiCad/kicad-doc #156: Capitalisation in document names

BrianSidebotham, 2015-09-15:
Assuming file systems are case sensitive there's a problem using capitlisation in the file names of the compiled documentation.

We must adhere to producing whatever the KiCad program is looking for. The issue is really down to the directory names which become targets in the build system, so the directory names need to be changed to suit what the KiCad programs are looking for.

e.g. `Pcbnew` should become `pcbnew`

nickoe, 2015-09-15:
Yes, I agree. I would be happy if we can go all lower case on file names. I am not sure why, but at some point @ciampix changed it to be capitalized. This was a long time ago, when it was mostly him working on the new docs. I don't remember if there were a specific reason for it, if it was technical it must have been related to the old makefile system.

ciampix, 2015-09-15:
Yes, it was a really ugly hack to avoid having conf files for docs and a single Makefile. You can rename dirs without problem. I will fix the Makefile just to have a second method to compile docs (for now). Then we can get rid of the Makefile altogether without regret...

nickoe, 2015-09-15:
@ciampix are we missing any features from the plain make files as of now? If so, what is it? If no, then I think it should be ok to just remove them now.

ciampix, 2015-09-15:
I am always insecure... yes please go on... anyway if the dir are all lovercase as the filenames, makefile should work anyway... I am really wondering at what I was thinking when I decided to create such camelcase craziness...

ciampix, 2015-09-16:
Found! I was sure capitalization was for something: an ugly hack for creating docinfo file for correct title on epub cover... and for some reason it does not work anymore: I see the KiCad logo but not the document title so this hack is not useful at all...
Anyway if you like I have ready a patch to lovercase all docs and dirs. If you prefer it in this way I can push it right now...

BrianSidebotham, 2015-09-16:
Please push now Marco, yes. Thank-you for the quick action and confirmation
of what the original problem was!

commit 160d24db (2015-09-16): Converted all docs and docs dirs to lowercase to avoid
camecase portability problems. Closes #156.
closed 2015-09-16 by ciampix