KiCad/kicad-doc #158: Changing to lowercase docs & dirs broke jenkins run

ciampix, 2015-09-16:
Changing to lowercase docs & dirs broke jenkins run. Presumably the script should be updated since it refers to the old CamelCase docs dir names like CvPcb. @nickoe could you see to this? I image the doc web pages should also be updated...

nickoe, 2015-09-16:
Yes I have just update the build script. Lets see how it goes.

http://ci.kicad-pcb.org/job/any-kicad-doc-head/552/

kinichiro, 2015-09-16:
Hi, @ciampix 
I saw your commit https://github.com/KiCad/kicad-doc/commit/160d24db3cb57ff40e72c05267b0343e1330cb7a and I noticed some faults.
Some files seems to renamed to other directories.
<pre>
...
 src/Pcbnew/images/icons/down.png → src/cvpcb/images/icons/down.png
 src/Pcbnew/images/icons/left.png → src/cvpcb/images/icons/left.png
...
 src/Pl_Editor/po/addendum.fr → src/cvpcb/po/addendum.fr
 src/Pl_Editor/po/addendum.it → src/cvpcb/po/addendum.it
...
</pre>

ciampix, 2015-09-16:
agh! seems that git done something not so "smart". Probably I have to do some smaller commits and not one gigantic... probably it is best to revert it and re do it one doc dir at a time...

ciampix, 2015-09-16:
one moment... I do not see any of the reported "confusion" of git. Here all renamed file are correct. Are you sure? Is see for example: src/Pcbnew/images/icons/down.png → src/pcbnew/images/icons/down.png

ciampix, 2015-09-16:
Checked on github web interface. It seems to me a github bug. Git reports it correctly...

nickoe, 2015-09-16:
I just breiefly looked at the jenkins log and it seem like some includes have not been updated. Please have a look at that, I am offline now.

kinichiro, 2015-09-16:
sorry, it seems a github displaying issue.

kinichiro, 2015-09-16:
in src/pcbnew/pcbnew.adoc, stiil remains case sensitive includes.
<pre>
include::Pcbnew_introduction.adoc[po4a]
include::Pcbnew_installation.adoc[po4a]
...
</pre>

ciampix, 2015-09-16:
Yes this is my fault... I'll commit a remedy now... sorry

kinichiro, 2015-09-16:
all the .po files should be refreshed since they still include case sensitive file name lines, like this.
<pre>
#. type: Plain text
#: Eeschema.adoc:23
#, no-wrap
...
</pre>

ciampix, 2015-09-16:
Well that is just a comment, not really an issue. Next time the source will be changed the po files will update automatically.
I'll check the compilation results. If all is ok I'll close this issue.

kinichiro, 2015-09-16:
Thanks. I see.
I compiled in my local PC, and verify all docs are good.
I think it is OK.
Thanks, again.

nickoe, 2015-09-16:
Yes, looks good to me, the warning level is back to normal.

closed 2015-09-16 by nickoe