KiCad/kicad-doc #160: Built documents installed in build path.

stambaughw, 2015-09-21:
Why does the build configuration install the built documentation in the build path rather than in the platform specific path defined by CMAKE_INSTALL_PREFIX?  I cannot do a `make install` and have the documentation files end up in the correct path.  I have to manually copy them from the build path over to the correct system path.  This should be more like the old documentation install configuration.  Is this by design or an oversight?

LordBlick, 2015-09-21:
Seems that it is by bad `DESTINATION` paths prefixes (`./`) in `CMakeModules/KiCadDocumentation.cmake` (`INSTALL`).

nickoe, 2015-09-21:
This is indeed an annoying issue for packagers. IIRC the current structure output by the build does not match the KiCad search paths very well. I have not had too much time to dig into it, feel free to give it a look. I have not included the docs in the windows builds yet, because of this.

LordBlick, 2015-09-21:
Try to remove all `./` in `DESTINATION` at `INSTALL` lines, then should be much better…

stambaughw, 2015-09-21:
I'll take a look at it when I get a chance.  I've had limited success
building the documentation on Debian testing unless I just build the
English docs so I haven't gotten to the install point just yet.

nickoe, 2015-09-21:
Please note that building with more jobs might not work properly (-jn make
option). At least I have not had much luck with it.

And an annoying issue when you need to rerun,
https://github.com/KiCad/kicad-doc/issues/73

nickoe, 2015-09-24:
@LordBlick Yes, that definitely seem to be the problem. I am trying to fix this. Expect a commit to fix this today.

@stambaughw FYI

stambaughw, 2015-09-24:
@Nick, I didn't test this patch but it should look something like the
attached patch.  I removed the pdf path for the pdf files since that's
where kicad expects to find them.  I left the html and epub paths in
tact.  I'm not really sure what we want to do about that since they are
not need by kicad.

diff --git a/CMakeLists.txt b/CMakeLists.txt
index 917d75b..3bf7cc4 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -125,6 +125,10 @@ if( GIT_FOUND )
     endif()
 endif()
 
+# Everything without leading / is relative to CMAKE_INSTALL_PREFIX.
+set( KICAD_DOC_INSTALL_PATH share/doc/kicad
+     CACHE PATH "Install path of KiCad documentation files relative to CMAKE_INSTALL_PREFIX." )
+
 # Set the language in the package name if the package only contains a single
 # language
 if( "${SINGLE_LANGUAGE}" STREQUAL "" )
diff --git a/CMakeModules/KiCadDocumentation.cmake b/CMakeModules/KiCadDocumentation.cmake
index 711c9b2..aeb4518 100644
--- a/CMakeModules/KiCadDocumentation.cmake
+++ b/CMakeModules/KiCadDocumentation.cmake
@@ -113,11 +113,14 @@ macro( KiCadDocumentation DOCNAME )
             add_dependencies( ${DOCNAME}_html_${LANGUAGE} ${DOCNAME}_translate_${LANGUAGE} )
             add_dependencies( ${DOCNAME} ${DOCNAME}_html_${LANGUAGE} )
 
-            install( FILES ${CMAKE_CURRENT_BINARY_DIR}/${LANGUAGE}/${DOCNAME}.html DESTINATION ./${LANGUAGE}/${DOCNAME}/html COMPONENT html-${LANGUAGE} )
-            install( DIRECTORY ${CMAKE_CURRENT_BINARY_DIR}/${LANGUAGE}/images DESTINATION ./${LANGUAGE}/${DOCNAME}/html COMPONENT html-${LANGUAGE})
+            install( FILES ${CMAKE_CURRENT_BINARY_DIR}/${LANGUAGE}/${DOCNAME}.html
+                     DESTINATION ${KICAD_DOC_INSTALL_PATH}/${LANGUAGE}/${DOCNAME}/html
+                     COMPONENT html-${LANGUAGE} )
+            install( DIRECTORY ${CMAKE_CURRENT_BINARY_DIR}/${LANGUAGE}/images
+                     DESTINATION ${KICAD_DOC_INSTALL_PATH}/${LANGUAGE}/${DOCNAME}/html
+                     COMPONENT html-${LANGUAGE} )
         endif()
 
-
         # PDF Generation
         list( FIND BUILD_FORMATS "pdf" PDF_BUILD )
         if( NOT "${PDF_BUILD}" EQUAL "-1" )
@@ -129,10 +132,13 @@ macro( KiCadDocumentation DOCNAME )
             add_dependencies( ${DOCNAME}_pdf_${LANGUAGE} ${DOCNAME}_translate_${LANGUAGE} )
             add_dependencies( ${DOCNAME} ${DOCNAME}_pdf_${LANGUAGE} )
 
-            install( FILES ${CMAKE_CURRENT_BINARY_DIR}/${LANGUAGE}/${DOCNAME}.pdf DESTINATION ./${LANGUAGE}/${DOCNAME}/pdf COMPONENT pdf-${LANGUAGE})
+            # PDF files are the default documentation for KiCad so they get installed
+            # in the root language path.
+            install( FILES ${CMAKE_CURRENT_BINARY_DIR}/${LANGUAGE}/${DOCNAME}.pdf
+                     DESTINATION ${KICAD_DOC_INSTALL_PATH}/${LANGUAGE}/${DOCNAME}
+                     COMPONENT pdf-${LANGUAGE} )
         endif()
 
-
         # EPUB Generation
         list( FIND BUILD_FORMATS "epub" EPUB_BUILD )
         if( NOT "${EPUB_BUILD}" EQUAL "-1" )
@@ -144,7 +150,9 @@ macro( KiCadDocumentation DOCNAME )
             add_dependencies( ${DOCNAME}_epub_${LANGUAGE} ${DOCNAME}_translate_${LANGUAGE} )
             add_dependencies( ${DOCNAME} ${DOCNAME}_epub_${LANGUAGE} )
 
-            install( FILES ${CMAKE_CURRENT_BINARY_DIR}/${LANGUAGE}/${DOCNAME}.epub DESTINATION ./${LANGUAGE}/${DOCNAME}/epub COMPONENT epub-${LANGUAGE})
+            install( FILES ${CMAKE_CURRENT_BINARY_DIR}/${LANGUAGE}/${DOCNAME}.epub
+                     DESTINATION ${KICAD_DOC_INSTALL_PATH}/${LANGUAGE}/${DOCNAME}/epub
+                     COMPONENT epub-${LANGUAGE} )
         endif()
     endforeach()

nickoe, 2015-09-24:
@stambaughw I actually already comitted a similar fix in https://github.com/KiCad/kicad-doc/commit/8739e5ee368104073165f0390385a8f63ce295e2

The html is supported by the kicad path serching stuff, but the epub is not. But it could be in the future. For now I think we might as well include all the formats. If a packager don't want all formats, he can just build the pdf one.

closed 2015-09-24 by nickoe