KiCad/kicad-doc #161: place of documents

kinichiro, 2015-09-23:
The place of KiCad documents in local computer was discussed in the ML and the bug tracker.
- https://lists.launchpad.net/kicad-developers/msg20504.html
- https://bugs.launchpad.net/kicad/4.0/+bug/1496049

I think it is good to describe about this 
in the chapter "More about KiCad documentation" of getting_started_in_kicad.

IMHO, it should describe about
- `KIHELPDIR`
- default location for linux, windows(32bit), windows(64bit) and Mac

nickoe, 2015-09-23:
I have added some paths for the systems, but I have not described the KIHELPDIR, I don't know anything about that. See ddb1a32.

kinichiro, 2015-09-24:
Thanks for your updating the doc.
It seems that KIHELPDIR is used only installation time.
I thought that KIHELPDIR is used when KiCad softwares search the path of docs.
It was my misunderstanding, sorry.

I think your updating is good, so I'll close this issue with this comment.

closed 2015-09-24 by kinichiro