KiCad/kicad-doc #163: po4a-updatepo for multi file document and full path is recorded in po

kinichiro, 2015-09-25:
I've found 2 issues around po4a-updatepo with cmake build.

1. It seems that `make (docname)_updatepo_(lang)` doesn't work well if doc is multi file.
   Old Makefile did po4a-updatepo like this.
<pre>
@po4a-updatepo -f asciidoc -v -M utf-8 -m $(DOCNAME).adoc $(foreach l,$(DOCCHAPTERS),-m $(l)) -p $@
</pre>
   Yes, it passed all the .adoc files with -m.
   But CMakeModules/KiCadDocumentation.cmake does 
<pre>
COMMAND ${PO4A_COMMAND}-updatepo -f asciidoc -v -M utf-8 -m ${CMAKE_CURRENT_SOURCE_DIR}/${DOCNAME}.adoc -p ${CMAKE_CURRENT_SOURCE_DIR}/po/${LANGUAGE}.po )
</pre>
   All .adoc files should be passed to po4a-update command with -m flag.

2. full path of .adoc is recorded in .po file
   I updated po by `make cvpcb_updatepo_ja`. 
   Then, comment line is like this, full path is recorded.
<pre>
#. type: Title =
#: /home/inoguchi/kicad/src/cvpcb/cvpcb.adoc:6
#, no-wrap
msgid "CvPcb"
</pre>
   Old Makefile just recorded .adoc filename only, not full path, and this was good.
   Can you cd to ${CMAKE_CURRENT_SOURCE_DIR} first,
   then po4a-update -m without full path ?

Thanks.

kinichiro, 2015-10-20:
@ciampix  Thanks,
utils/old-build-scripts/update-po-all.sh added by commit https://github.com/KiCad/kicad-doc/commit/8e9cd00fb5dee4b96b1798ad7bc51c4a8214e9e2 is useful for me.

ciampix, 2015-10-20:
I just want to maintain old working script until we are sure of cmake working perfectly. We should warn directly @Caerbannog about these issues...
In the future probably we will have to periodically update all po files to let translators concentrate only on their work. If we will manage to arrange a web translation service we will have to update automatically all po files too.

yoneken, 2015-10-20:
We (i18n users) don't need automatically update..
We prefer ambiguous i18n words to correct english words.

A web translation service is more better.

ciampix, 2015-12-29:
@yoneken I do not really understand what you mean here. Automatic update is useful because (English) source files are not _static_ as doc authors continue to change the doc sources. This is due to doc corrections and/or improvements and is not avoidable.

kinichiro, 2016-05-05:
This issue is resolved by PR #353 .

closed 2016-05-05 by kinichiro