KiCad/kicad-doc #166: eliminate multi byte char from adoc

kinichiro, 2015-09-27:
IMHO, adoc should not use multi byte char
if it is able to express by single byte char.

I think these could be eliminated from adoc.
- multi byte dash (– -> -)
- multi byte single quotation (’ and ‘ -> ')
- multi byte double quotation (“ and ” -> ")

And I also think these could be permitted.
- →
- …
- °

I listed up the adoc lines containing multi byte char here.
https://gist.github.com/kinichiro/9f3919491644e6f0f76d

How do you think about this ?

Thanks

nickoe, 2015-09-27:
What is the reason for this, even though it is called asciidoc, it supports UTF-8. A2x just needs the lang attribute set.

kinichiro, 2015-09-27:
Because some multi byte char seems give a bad influence on some editor.

For example, editing eeschema_general_top_toolbar.adoc with vi.
You can see multi byte '-' is displayed '?'.
[image]

Editing pcbnew_routing.adoc with vi.
You can see some '?', but they should be single white space.
[image]

I don't mean all multi byte char should be eliminated.
But, I think we should not use it as far as possible for avoiding troubles.

ciampix, 2015-09-27:
I agree 100%.

LordBlick, 2015-09-28:
Tool should be reported as buggy, not using std utf8…

nickoe, 2015-09-30:
@kinichiro That is simply because your terminal or font does not support it properly, please use a proper environment. Something that supports UTF-8. But that part is indeed a bit funny, but the problem is that someone who wrote that has inserted non breaking space. If you inspect the positions with a hex editor you will see it as C2 A0 in the pcbnew doc. And an en dash in the eeschema doc.

I suspect theese codes actually appeared from the odt to adoc conversion.

I think it does not hurt keeping them, but I guess it wont really hurt ot fix them either. if you feel like fixing these, I am ok with accepting a pull request on this.

I guess using the non multibyte chars for theese special chars will make it easier to debug if we find funky build errors too. But the last part is only a suspicion.

kinichiro, 2015-10-03:
@nickoe 
>I suspect theese codes actually appeared from the odt to adoc conversion.
I think so, too.
And I would like to remove unneeded multi byte chars as far as I can for avoiding unexpected trouble.

nickoe, 2015-10-03:
I will just close this since some commits to address this issue has been merged.

closed 2015-10-03 by nickoe

ciampix, 2015-10-03:
Thanks @nickoe