KiCad/kicad-doc #167: Support parallel builds

onitake, 2015-09-28:
The dependencies for each target seem to be incomplete, causing parallel builds to fail.

It would be great to have this fixed before KiCad 4.0 is released, as I'd like to integrate documentation compilation into the Debian package building process. Without parallel make support, build time is increased dramatically on a multi-core system.

nickoe, 2015-09-28:
Yes indeed this is an annoying problem. There are multiple dependency issues with the cmake script as is. Feel free to analyse this issue further and possibly propose a fix for it. Parallel builds are not highest on priority list at the moment.

BrianSidebotham, 2015-09-28:
Also see issue #73 which is related of course...

onitake, 2015-09-29:
Thank you, parallel builds are now working for me. Tested with -j4.

Now there's only the issue that `make install` will rebuild all documentation from scratch. With the faster parallel build, this is much less of a bother though.

nickoe, 2015-09-30:
@BrianSidebotham It seems that this issue is not completely fixed. I just tried this twice with -j4 on my machine, see this gist: https://gist.github.com/nickoe/72484fccd8219c6da105

onitake, 2015-09-30:
Huh. Maybe it's something else? I hit several problems like this until I had all required dependencies installed. Like, the Japanese fonts for example.

Try running the failing command manually and check the output, that's what helped me at least.

nickoe, 2015-09-30:
@onitake The command seem to work fine, and also --- it builds fine with -j1 but it will of couse take some more time.

BrianSidebotham, 2015-09-30:
@nickoe no problem, with parallel jobs this is how things can go. In the root CMakeLists.txt please uncomment the `add_a2x_option( --verbose )` at https://github.com/KiCad/kicad-doc/blob/master/CMakeLists.txt#L40 so that we get more information from a2x as to what goes wrong. Otherwise it's a bit of a shot in the dark because the errors don't filter through to us.

nickoe, 2015-09-30:
@BrianSidebotham Ok, I have now generated a new and more verbose log at https://gist.github.com/nickoe/81c68869d12a5595f722. I hope you can make some sense of the issue here. I know this kind of dependency issues can be hard to debug.

commit 07684c4e (2015-09-28): Merge pull request #168 from BrianSidebotham/dependencies

Fix parallel build issue ( Fixes #167 )
commit 412d72b0 (2015-09-28): Fix parallel build issue ( Fixes #167 )
commit 880ceb09 (2015-10-01): Merge pull request #175 from BrianSidebotham/parallelfix

Sure up multiple ax2 runs for parallel jobs in the same folder (Fixes #167)
closed 2015-10-01 by nickoe