KiCad/kicad-doc #170: Installed HTML help contains images for other languages

onitake, 2015-09-29:
I'm not sure if this is related to #168, but images for other languages "bleed" into each HTML build result.

For example, if I build the English documentation and `make install` it, I will get a `/usr/share/doc/kicad/help/en` with a subdirectory `images`. Inside, there are the images for the English HTML files, but also `it` and `ja` subdirectories. These contain images for the respective languages.

Is this intended, or a result of my parallel build (with `make -j4`)?

nickoe, 2015-09-29:
This is indeed a bug. I appeared after I changed it to install like kicad wanted the help files. I intend to fix this.

BrianSidebotham, 2015-10-02:
I'm not sure this is new. I think this has always been the case although not desirable.

At the moment it's one of the things that adds to the long packaging time

onitake, 2015-10-02:
It's also strange that some languages are not included. I get en, it, pl and ja subfolders for each language, but never fr or nl.

nickoe, 2015-10-02:
Mmm, it seems that https://github.com/KiCad/kicad-doc/commit/f52c2c89c274d079062fe7a6e8234459e099a193 does not fix this issue at all.

ciampix, 2015-12-18:
Is this bug still present? If not, may I close this? If yes, how can I reproduce it?

nickoe, 2015-12-19:
@ciampix I don't remember if I really managed to fix this properly. It is solely a matter of testing it.

ciampix, 2015-12-19:
ok
cmake ..
make make getting_started_in_kicad_html_en
... so long so good...
sudo make install getting_started_in_kicad_html_en
install _ALL_ the images... apparently the bug is still there...

closed 2016-12-18 by nickoe