KiCad/kicad-doc #172: Mac Dependencies?

adamwolf, 2015-10-01:
Has anyone tried to build this on OS X recently?  If so, any tips?

nickoe, 2015-10-01:
I have not personally tried it, I don't have access to a woring OS X system, and never had good luck with it in virtualbox.

But I asked someone to try it out and he said that he just ran cmake and make, but not make install. I guess that it means it basically works.

So... @adamwolf what is your problem if you have any? The dependencies are listed in the README.adoc in the root of the repo.

BrianSidebotham, 2015-10-01:
Just making myself aware of this issue so I get the updates. Please test on OSX Adam. I've not tested it at all.

If you can get it working and have some additions to the readme's for the dependencies or any OSX specific information please do a pull request so it can be included - it'll be greatly appreciated!

adamwolf, 2015-10-01:
I really want to get this working on OS X, but it won't happen until after
the stable release.

I'm certain it works if you get the dependencies right, but some of them
are really, really, esoteric on Mac, and not supported by their maintainers
on OS X--I don't have my notes with me but if i have some downtime soon
I'll copy them over so I have some place to start after the stable release.

For now, I'm just copying over from a Linux Jenkins slave.

BrianSidebotham, 2015-10-01:
If it helps - I don't really see much value in building on all platforms these days. One way to tackle it is to look at using Docker. There is a Docker script already, although I'm not sure it's that up-to-date: https://github.com/KiCad/kicad-doc/tree/master/utils/docker

Moving forward I see containers being the obvious way to build anything on any platform with much more ease. The dependencies on Windows for this are also horrendous!

nickoe, 2015-10-01:
Anyway, I intend to keep the cpack generate docs on docs.kicad-pcb.org, such that it is easy for packagers to just grab those and also save the time rebuilding the docs.

adamwolf, 2015-10-01:
Great.  I will adjust the instructions on my builder stuff to say "folks,
go grab a recentish doc set from http://wherever" and probably call it a
day... life's too short to chase down these dependencies :)

evanshultz, 2016-06-22:
@ciampix Reading the above thread, it seems there's a plan and nothing is alarming anyone. Can this issue be closed?

ciampix, 2016-06-22:
Yes, this PR is almost 1 year old, and anyway, I am planning to pass all build chain to asciidoctor that has almost no dependencies at all.

closed 2016-06-22 by ciampix