KiCad/kicad-doc #183: update Getting Started in KiCad for 4.0

kinichiro, 2015-10-10:
Some part of GSiK should be updated according to 4.0.
Are there any other part that should be updated ?
Please comment here.

## Unification of words
 - [ ] EESchema -> Eeschema
 - [ ] PCBnew -> Pcbnew

## 3. Draw electronic schematics
 - [ ] 19. microcontrollers library -> microchip_pic12mcu.lib
 - [ ] 19. PIC12C508A -> PIC12C508A-I/SN
 - [ ] 22., 28., 30., 31., 37., 39., 41.,  update image
 - [ ] 47. cvpcb is kicked from Eeschema not KiCad project manager
 - [ ] 50. update footprint name (3PIN_6mm, DIP-8_300, ...)
 - [ ] 52. tute1.net -> tutorial1.net
 - [ ] 56. Note section,  /usr/lib/kicad/plugins/ should update

## 4. Layout printed circuit boards
 - [ ] 2. Tute1 -> Tutorial1
 - [ ] 5. tute1.net -> tutorial1.net
 - [ ] 4.1 3. KiCad Layer Name should be updated

## 5. Make schematic components in KiCad
 - [ ] 1., 8., 9. demo1 -> tutorial1

## 6. Make schematic components with quicklib
 - [ ] 4., 5. demo1 -> tutorial1
 - [ ] 6. myOwnLib.lib -> myLib.lib

## 7. Make component footprints
 - [ ] .mod -> .kicad_mod
 - [ ] footprint library name "connect" should be updated

## 8. Note about portability of KiCad project files
 - [ ] .mod -> .kicad_mod
 - [ ] image of project directory

ciampix, 2015-10-12:
Regarding this:

    EESchema -> Eeschema
    PCBnew -> Pcbnew

I simply converted the OpenOffice doc. I really do not know if that CamelCase is of some meaning or importance if any...

kinichiro, 2015-10-12:
About 7. Make component footprints

Current tutorial saves the new footprint in the system wide library, under $KISYSMOD.
This should change to the project library, like ${KIPRJMOD}/myfootprint.pretty.
And footprint of MYCONN3 is stored in tutorial1/myfootprint.pretty/MYCONN3.kicad_mod.

kinichiro, 2015-10-13:
Add NOTE block in the section of 4.2 (Freerouter)

Freerouter is no more a web application,
but now it is a standalone java application that should be built by user.
 - https://bugs.launchpad.net/kicad/+bug/1290022
 - http://www.freerouting.net/fen/viewtopic.php?f=4&t=255
 - https://github.com/nikropht/FreeRouting

GSiK should notify to user about this in 4.2.
<pre>
NOTE:
Freerouter is a open source java application,
and it is needed to build by yourself to use with KiCad.
Source code of Freerouter can be found on this site:
https://github.com/nikropht/FreeRouting
</pre>

Any comments are appreciated :-)

ciampix, 2015-10-15:
Very well, go ahead!

kinichiro, 2015-10-15:
Organize table of contents.

- sections of "Make scheematic components" put into one chapter.
- move up "Forward annotation" chapter after "4. Layouter PCB".

<pre>
1. Introduction to KiCad
2. KiCad Workflow
3. Draw electronic schematics
4. Layout printed circuit boards
5. Forward annotation in KiCad (was 9.)
6. Make schematic components in KiCad (was 5.)
 6.1. Make schematic components with quicklib (was 6.)
 6.2. Make a high pin count schematic component (was 6.1)
7. Make component footprints
8. Note about portability of KiCad project files
9. More about KiCad documentation (was 10.)
</pre>

Then,
Chapter 3., 4. and 5. explains whole of the PCB design work,
Chapter 6. and 7. explains how to make components and footprint,
Chapter 8. explains portability of artifacts.

kinichiro, 2015-10-15:
@ciampix ok, I'll start fixing this issue.

kinichiro, 2015-10-17:
Thanks for merging the PR.
I will close this issue.

closed 2015-10-17 by kinichiro