KiCad/kicad-doc #209: fail creating .po file

AntonioMR, 2015-11-16:
I'd like to translate the doc files into spanish but I'm having problems during the .po file creating sentence
I followed the steps, i've installed all the dependences, i've downloaded the src files, created the buid directory and execute the make command to build the english version of the documents.
But when i run the "make po/es.po" sentence inside the kicad-doc/src/eeschema/ directory, i obtain the next error msg:
"there is not any rule to build the object"
Any idea about how to continue?

kinichiro, 2015-11-17:
Hi,
You can try this.

cd src/＜your target doc＞
touch *.adoc
make -f ../../utils/old-build-scripts/Makefile po/es.po

Regards

ciampix, 2015-11-17:
Correct, thanks!

AntonioMR, 2015-11-17:
It works!

Thanks a lot.

timofonic, 2015-11-17:
Are you doing to publish your efforts early in a GitHub fork?

I would like to try to help translating this. I'm trying it with KiCad
itself and right now I'm damn busy. But I would love to contribute
effectively one of these days.

AntonioMR, 2015-11-18:
yes, I've already done a fork. I'm going to do this stuff stealing sleeping hours, so any help will be wellcome

ciampix, 2015-11-21:
So this seems resolved. I am closing it.

closed 2015-11-21 by ciampix