KiCad/kicad-doc #210: bom2csv Plugin Error

boris81, 2015-11-18:
This is regarding *Getting Started in KiCad* document, page 19, step 56.  If done as specified, the *bom2csv* plugin will not work.  

The following code should be added in the note section:
>*sudo apt-get install xsltproc*

Once installed, then the plugin should be downloaded:
>wget https://raw.githubusercontent.com/KiCad/kicad-source-mirror/master/eeschema/plugins/bom2csv.xsl

ciampix, 2015-11-18:
Good catch! That could be certainly added for Debian/Linux or similar and I am already preparing a patch but I wonder how users can do with Mac OS X or Windows...

nickoe, 2015-11-18:
The xsl files are includes in the windows installer.

kinichiro, 2015-11-18:
On Windows, xsltproc.exe is placed under the directory C:\Program Files\KiCad\bin .
I checked this with kicad-product-4.0.0_rc2-x86_64.exe.
OS X, I have no idea since I don't have it.

ciampix, 2015-11-18:
So, in Linux, it could not be mentioned since it should be just a matter of package dependencies that should already manage to have xsltproc installed with KiCad, right?
If the Mac OS X already contains the xlstproc executable, this note could be avoided at all...

Caerbannog, 2015-11-19:
@ciampix Linux package dependencies should indeed install it. But someone will probably have the problem one day: bad package, install from source, ...

kinichiro, 2015-11-19:
IMHO, how about mentioning that xsltproc is the OSS and users might install by themselves.
Like description about Freerouter in GSiK.
And add the installation example for Ubuntu only.

nickoe, 2015-11-28:
Does this need some changes to the docs? I will soon tag the docs for the release, so any changed that needs to be included shall be made very very soon.

boris81, 2015-11-30:
@nickoe Please add the following information on the document:

>*sudo apt-get install xsltproc*

It should go on page 19, step 56, note section, right before *wget https://raw.githubusercontent.com/KiCad/kicad-source-mirror/master/eeschema/plugins/bom2csv.*

evanshultz, 2016-06-16:
I'd like to update the doc and close this issue, but I'm not sure what to do. Can someone help?

It seems to me this is broken down by OS.

*Linux*
Since this will (apparently) be handled with a package dependency, do we or do we not need to include the code above? Maybe it's best to note this dependency and include the code, so that we've provided all the important info for users if they have a problem?

*Windows*
On Win10 x64 and Win7 x64, I find bom2csv.xsl at C:\Program Files\KiCad\bin\scripting\plugins. I don't know about other versions of Windows, but that's the vast majority.

*macOS*
Nobody above knows about macOS, and I don't have the computer to test it.

boris81, 2016-06-23:
I think we are good to go.  

Regarding _macOS_, you can always place some sort of placeholder, such as TBD (To Be Determine) in there.  Can you?

commit d60dd286 (2016-07-01): Added more intall options for xlstproc for GSIK manual, closes #210
closed 2016-07-01 by ciampix