KiCad/kicad-doc #221: Build (all) is currently failing

tijuca, 2015-12-08:
I'm trying to build the current HEAD in master, but the build is failing early
7d9642f Updated Italian translation in eeschema manual

I have done:
```
$ git clone https://github.com/KiCad/kicad-doc.git
...
$ cd kicad-doc/
$ mkdir build
$ cd build/
$ cmake ..
$ make
Built target cvpcb_translate_footprint_libraries_tables_en
Built target cvpcb_translate_en
Built target cvpcb_html_en
a2x: ERROR: "dblatex" -t pdf -p "/etc/asciidoc/dblatex/asciidoc-dblatex.xsl" -s "/etc/asciidoc/dblatex/asciidoc-dblatex.sty" -P latex.output.revhistory=0 -P doc.publisher.show=0 -s /home/carsten/gitprojects/kicad-doc/CMakeSupport/pdf-cover-dblatex.sty -b xetex -p /home/carsten/gitprojects/kicad-doc/src/cvpcb/../../xsl/dblatex-pdf-ja.xsl -o /home/carsten/gitprojects/kicad-doc/build/src/cvpcb/en/cvpcb.pdf  "/home/carsten/gitprojects/kicad-doc/build/src/cvpcb/en/cvpcb.xml" returned non-zero exit status 1
src/cvpcb/CMakeFiles/cvpcb_pdf_en.dir/build.make:57: recipe for target 'src/cvpcb/CMakeFiles/cvpcb_pdf_en' failed
make[2]: *** [src/cvpcb/CMakeFiles/cvpcb_pdf_en] Error 1
CMakeFiles/Makefile2:599: recipe for target 'src/cvpcb/CMakeFiles/cvpcb_pdf_en.dir/all' failed
make[1]: *** [src/cvpcb/CMakeFiles/cvpcb_pdf_en.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
$ dblatex --version
dblatex version 0.3.7-1
$ a2x --version
a2x 8.6.9
```
Also all the files from the dblatex call are in place, what I'm currently missing?

ciampix, 2015-12-08:
I doubt that the Italian translation have something to do with this since your local make fails do create the English pdf...

tijuca, 2015-12-08:
O.k. figured out what's going wrong here.
The package ``texlive-lang-dutch`` isn't alive after Debian Jessie (I'm running testing/stretch), so the needed dependency chain could't be solved. The Dutch language for tex lives now in the package ``texlive-lang-european``
--> https://packages.debian.org/jessie/texlive-lang-european
The package ``texlive-lang-dutch`` is a transitional package in Jessie so it would be good to update the HowTo in the Wiki.

As KiCad is growing in the future a other possibility is to just use the package ``texlive-lang-all`` in the HowTo section, it will pull all the available texlive packages.

ciampix, 2015-12-08:
Updated README thanks!

commit 09d0260b (2015-12-08): Updated in README the build dependencies. Closes #221.
closed 2015-12-08 by ciampix