KiCad/kicad-doc #224: GSIK: Section 1; cvpcb no-longer standalone

bobc, 2015-12-15:
cvpcb is no longer a standalone tool. I suggest that it's function "footprint selector" be added to eeschema row, and cvpcb row removed.

kinichiro, 2015-12-17:
absolutely.
cvpcb is no more a standalone program,
but it is still described as one program in some documents
since it is transition period of changing KiCad.
not only that, it has its own independent reference manual, still.
it needs more time to fix these issues.

bobc, 2015-12-17:
Surely we should change those other documents as well then :puzzled:

evanshultz, 2016-06-19:
Six months later it appears there's no change. I'm happy to do this and will start on it shortly.

evanshultz, 2016-06-27:
The patches for GSIK (and also KiCad) are merged. @ciampix you may close this issue if you like, as the purpose of this issue is addressed and I'll merge the actual CvPcb documentation into Eeschema when I get there.

ciampix, 2016-06-27:
yes of course, thanks!

closed 2016-06-27 by ciampix