KiCad/kicad-doc #225: GSIK: 2.1 workflow diag could be improved

bobc, 2015-12-15:
The work-flow diagram is really useful, but seems to have some issues:
1) annotate and generate netlist is not part of pcbnew as implied
2) Footprint Editor is not solely part of pcbnew as implied (perhaps minor)
3) the diagram is scrunched up in places. I realise we don't want it to become too big, but the readability suffers. If the work-flow really is a two stage process perhaps the diagram can be split into two? 
4) Viewing Gerber files could be shown as a third stage process

ciampix, 2015-12-16:
1) I do not think that that graph imply anything ... it is a work-flow of actions performed inside the various programs...
2) ditto
3/4) the meaning of the graph is to give an overview of the whole process. To cut it into two or three pieces it could be typographically cleaner but I am afraid not clearer from a user prospective
But these are just my thoughts, as I am thinking out loud; I am not sure of anything. I would like to hear the opinion of others before changing this.... @nickoe @Caerbannog @kinichiro ?

kinichiro, 2015-12-17:
Hi,

1) Indeed. 
   "Annotate components" should be placed before "Perform Rule Check"(ERC)
   "Generate Netlist" should be placed next to the bottom circle of Eeschema workflow.

2) This is work-flow as @ciampix mentioned, and blue area indicates drawing schematic work,
   green area indicates designing PCB, and red area indicates viewing gerber files.
   So, I feel it is OK.
   There are Library Editor on blue area and Footprint Editor on green area,
   they are not a Eeschema nor Pcbnew, but I think it is no problem.

3),4) I would like to keep this work-flow diagram in one.
   I feel this work-flow is good design to show the whole work with KiCad.
   But I agree that the readability suffers, since characters are crushed in the letter size printing.
   Adjusting the total layout of boxes, and using bigger fonts will resolve the issue, doesn't it ?

Regards,

bobc, 2015-12-17:
"it is a work-flow of actions performed inside the various programs..." 

well that is my point, it shows actions performed in the wrong programs. That is obvious to experienced users, but confusing for new users.

ciampix, 2015-12-19:
See if my improvement is good enough so I pull it and update png accordingly...

nickoe, 2015-12-19:
I see what you mean there. But I also think that the sheer size of the diagram is a problem. Maybe it is better do divide it into a few more parts, with the first one being an overview, and the seoncs detailing the flow in each app?

kinichiro, 2015-12-19:
@ciampix , I think your update is OK.
To see .dia file, Dia application is needed. So I attach png version below.
[image]

ciampix, 2015-12-21:
Reopened since the commit fixed just one topic.

ciampix, 2015-12-23:
Re-though ... if someone can suggest a way (means: clone,create 2/3 example image & ask for a pull) to resolve it, well. Otherwise I am closing it in a few days...

kinichiro, 2015-12-24:
> 2) Footprint Editor is not solely part of pcbnew as implied (perhaps minor)

Would you like to add rectangle areas for Library Editor, Footprint Editor and CvPCB ?
I think these rectangle areas could help to indicate running another program.
Here is sample.
[image]

ciampix, 2015-12-24:
Oh great!!! Can you please ask for a pull?

kinichiro, 2015-12-24:
sorry, I had mistaken to submit PR, take a look at #242, please.

ciampix, 2015-12-24:
No problem I think is great!

ciampix, 2016-06-14:
Closing since #242 closed it.

commit 55678ad4 (2015-12-21): Merge pull request #229 from ciampix/master

Improved flowchart dia, fix #225
commit a83df0ac (2015-12-21): Improved flowchart dia, fix #225
closed 2016-06-14 by ciampix