KiCad/kicad-doc #226: GSIK: shortcut keys could benefit from graphic

bobc, 2015-12-15:
Some users have found the specification of shortcuts confusing e.g. "pressing the Place component shortcut key (a). "

This could be improved by replacing the text with a graphic icon, e.g something like [image]
Some small and open source icons would need to be generated or found.

It might also be useful to have a summary of the shortcut keys used in the GSIK in an Appendix.

ciampix, 2015-12-16:
Please mind that shortcut keys are not the same for every language. Adding icons could be nice but we should supply all keys icons for every language that could possibly use those ...

bobc, 2015-12-16:
All the shortcut keys I looked at are the same for each language, so I don't understand your comment. However, there are a limited number of keys, I don't see a problem providing graphics for each language.

Can you tell me where the translation tables for shortcut keys are, I will try to generate icons programatically.

ciampix, 2015-12-16:
You are right of course. I was erroneously thinking about menu hotkeys...

kinichiro, 2015-12-17:
Hi,
Here are just some my thoughts.

- shortcut key graphic icon
  GSiK already has many program icons in it,
  and I feel we don't want to increase them any more.
  I would like to read the text, not viewing graphical icons.
  Instead of icons, how about using '[' and ']' with bold.
  e.g. "pressing the Place component shortcut key **[a]**. "

- summary of shortcut keys list
  shortcut keys can be redefined by users, so they are changed and customized.
  How about mentioning about '?' key more strong ?
  In the note of seciton 3.1 item 5 tells about '?' key, but it is weak.
  Adding new section, say, "2.3 Hotkeys" or something with screenshot
  will help user to notice how to get the list.
  IMHO, GSiK is not a reference manual but a guidance text, so it should describe "how to".

- shortcut key -> hotkey
  BTW, I think that "shortcut key" should be replaced with "hotkey".

Regards.

bobc, 2015-12-17:
Icons are bad? Really? Instead of making a box with ASCII-art, it is surely better to make a real graphic. I am happy to provide the graphics, and make a pull request.

If you put "[a]", you then need to add more text to explain that means type the "a" key, and type '[', 'a', ']' keys. That might be obvious to us, but anything that simplifies things for people who are newbies is surely a useful improvement.

I would suggest just summarising the hotkeys used in the tutorial, with a note that ? shows the current list. Obviously if the user has redefined keys, he will know that...

kinichiro, 2015-12-18:
IMHO

- we should add explanations about difference between hotkey and acceleration key in chapter 2

They are not recognized by users.

nickoe, 2015-12-19:
I completely agree with @kinichiro , it does not really add any value to add the keys a graphics. Writing them in round or square brackets, I don't really care, I think they work ok, both of them. And you you write "on the hotkey [a]" then it is clear that it is only one key.

ciampix, 2015-12-24:
Well... to be honest, graphics aesthetically add value but what @kinichiro and @nickoe are saying is that the message is just the same. Because of this the work in adding graphics and more size to the docs could be near worthless and have some downside also...

kinichiro, 2015-12-25:
Anyway, I thought that the explanation about accelerator key and hotkey is needed.
I made a draft and screenshots for the explanation.
I think this explanation should be placed in chapter 2., 
since this must be explained before using eeschema.
Any comments are welcome.

I believe descriptions should be easy and short, but this draft is too long a bit, I know.
I understood that explanation about accelerator key and hotkey is difficult.
And I think there are some exceptions.
- According to my draft, [x] on pcbnew should begin track immediately
- Alt + F1 is accelerator, but it immediately zoom

I am glad if someone provides better one than mine.

Please refer to https://bugs.launchpad.net/kicad/+bug/1492561 for
discussion about accelerator key and hotkey.

-----

2.3 accelerator key and hotkey

KiCad has two kinds of shortcut keys, **accelerator key** and **hotkey**.
They are resembled but a little bit different.

* accelerator key
  The accelerator key have the same behavior as the clicking on a menu or on a icon on the right toolbar.
  The accelerator key just enables the function. It does not work until mouse left clicking.
  The accelerator keys are showed on the right side of menu pane.
[image]

* hotkey
  The hotkey equals to "accelerator key" + "mouse left clicking".
  The hotkey begins its function immediately at the current cursor location where the key is typed.
  Users can edit the assignment of the hotkeys from Preferences menu.
  The hotkeys list is displayed by typing **[?]**.
[image]

For example, in case of "Place wire".

Type "Shift + W", cursor shape is changed. 
Next, left click the place that you want to begin, then you can draw a wire. 
This is the behavior of accelerator key.

Type **[w]**, drawing starts immediately from the cursor location where you type **[w]**.
This is the behavior of hotkey.

[Note]
In this document, hotkeys are expressed with brackets like this, **[a]**.
If you see **[a]**, just type the key of "a" on the keyboards.

-----

Regards.

evanshultz, 2016-06-22:
Quick update: Patches landed for the text above describing accelerator keys and hotkeys. @ciampix will investigate improved ways to notate key presses within asciidoc.