KiCad/kicad-doc #230: Make instructions not recent

transistorgrab, 2015-12-21:
Hi,

the instructions to generate a *.po file are not current.
 kicad-doc/translation_instructions.adoc 
states to use make, but without a Makefile make will do nothing...

Best Regards.

ciampix, 2015-12-21:
See if it works now...

MarcusFecit, 2015-12-23:
It does.
from your kicad-doc root
mkdir build
cd build
cmake ../

This will create the Makefile.

ciampix, 2015-12-23:
Check if the translation instructions are fixed now. If there are no problems I will close this bug in a few days...

transistorgrab, 2015-12-23:
Either I'm not doing it right or something does not work:
```
 -=# [~/mounts/Data/Projects/kicad-doc/build] #=- 
$ ls ../src/eeschema/po/
addendum.de  addendum.fr  addendum.ja  addendum.template  fr.po  ja.po
addendum.es  addendum.it  addendum.pl  es.po              it.po  pl.po
 -=# [~/mounts/Data/Projects/kicad-doc/build] #=- 
$ cat ../src/eeschema/po/addendum.de 
PO4A-HEADER: mode=after; position=^\[\[contributors\]\]; beginboundary=\[\[
[[translation]]
//Translators translate the word just below and delete this comment
*Übersetzung*

//Translators put your names below here in the addendum file
André S. <ansc.de@gmail.com>, 2015

 -=# [~/mounts/Data/Projects/kicad-doc/build] #=- 
$ cmake ..
-- Configuring done
-- Generating done
-- Build files have been written to: /home/andre/mounts/Data/Projects/kicad-doc/build
 -=# [~/mounts/Data/Projects/kicad-doc/build] #=- 
$ make eeschema_update_de
make: *** No rule to make target 'eeschema_update_de'.  Schluss.
 -=# [~/mounts/Data/Projects/kicad-doc/build] #=- 
$ ls
CMakeCache.txt  cmake_install.cmake  CPackSourceConfig.cmake  src
CMakeFiles      CPackConfig.cmake    Makefile
 -=# [~/mounts/Data/Projects/kicad-doc/build] #=- 
$ ls -lh Makefile 
-rwxrwxrwx 1 andre andre 243K Dez 23 23:16 Makefile
```

ciampix, 2015-12-23:
What's not clear in the passage " ... and delete this comment"?

MarcusFecit, 2015-12-24:
You have to run cmake ../
 *after* having created your addendum.de and de.po

transistorgrab, 2015-12-24:
Ok what I did after your comments:
I removed the comment line from the addendum.de file (I thought removing comments is optional…)
```
$ cat ../src/eeschema/po/addendum.de
PO4A-HEADER: mode=after; position=^\[\[contributors\]\]; beginboundary=\[\[
[[translation]]
*Übersetzung*

//Translators put your names below here in the addendum file
André S. <ansc.de@gmail.com>, 2015
```
then I deleted the 'build' directory and followed the steps from the translation_instructions:
```
 -=# [~/mounts/Data/Projects/kicad-doc] #=- 
$ rm -rf build/
 -=# [~/mounts/Data/Projects/kicad-doc] #=- 
$ mkdir  build
 -=# [~/mounts/Data/Projects/kicad-doc] #=- 
$ cd build/
 -=# [~/mounts/Data/Projects/kicad-doc/build] #=- 
$ cmake ../
-- Found DBLATEX: /usr/bin/dblatex (found version "0.3.5") 
-- Found PO4A: /usr/bin/po4a (found version "po4a Version 0.46
von Martin Quinson und Denis Barbier.

Copyright (C) 2002, 2003, 2004 Software in the Public Interest, Inc.
Dies ist freie Software; schauen Sie in den Quellcode für weitere
Bedingungen. Es gibt KEINE Gewährleistung, nicht einmal für die
Marktreife oder für die Verwendung für einen bestimmtem Zweck.") 
-- Found ASCIIDOC: /usr/bin/asciidoc (found version "8.6.9") 
-- Found A2X: /usr/bin/a2x (found version "8.6.9") 
-- Found Git: /usr/bin/git (found version "2.5.0") 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/andre/mounts/Data/Projects/kicad-doc/build
 -=# [~/mounts/Data/Projects/kicad-doc/build] #=- 
$ make eeschema_update_de
make: *** No rule to make target 'eeschema_update_de'.  Schluss.
```
Still no success. Any hints? The Makefile is about 243K big, but it contains _NO_ single line with 'update_' anywhere.
The last comment from MarcusFecit makes me wonder: do I need to create an empty de.po file? The translation_instructions seem to suggest that running the 'make' instruction will create that file. 

```
$ grep update_ Makefile
```
gives no results, but there are some results for this grep (though no '_de' is in the file):
```
$ grep eeschema_updatepo_ Makefile
# Target rules for targets named eeschema_updatepo_all
eeschema_updatepo_all: cmake_check_build_system
	$(MAKE) -f CMakeFiles/Makefile2 eeschema_updatepo_all
.PHONY : eeschema_updatepo_all
eeschema_updatepo_all/fast:
	$(MAKE) -f src/eeschema/CMakeFiles/eeschema_updatepo_all.dir/build.make src/eeschema/CMakeFiles/eeschema_updatepo_all.dir/build
.PHONY : eeschema_updatepo_all/fast
# Target rules for targets named eeschema_updatepo_es
eeschema_updatepo_es: cmake_check_build_system
	$(MAKE) -f CMakeFiles/Makefile2 eeschema_updatepo_es
.PHONY : eeschema_updatepo_es
[…]
.PHONY : eeschema_updatepo_pl/fast
	@echo "... eeschema_updatepo_es"
	@echo "... eeschema_updatepo_fr"
	@echo "... eeschema_updatepo_ja"
	@echo "... eeschema_updatepo_it"
	@echo "... eeschema_updatepo_pl"
	@echo "... eeschema_updatepo_all"
```

ciampix, 2015-12-25:
You are right. Cmake po build is definitely broken.
Try this:

rm -rf build/*
cd src/eeschema
make -f ../../utils/old-build-scripts/Makefile po/de.po

ciampix, 2015-12-25:
Modified translation instructions accordingly...

kinichiro, 2016-05-06:
PR #353 fixed this issue.

<pre>
$ mkdir build
$ cd build
$ cmake ..
...
$
$ make eeschema_updatepo_de
Scanning dependencies of target eeschema_updatepo_de
Parse input files...
done.
Updating po/de.po:
.............................................................................................................................................. done.
po/de.po: 1125 translated messages, 292 untranslated messages.
Built target eeschema_updatepo_de
$
</pre>

ciampix, 2016-05-06:
Yes thanks then I close this issue.

ciampix, 2016-05-06:
... wait a minute, I have to update the translation instructions ...

commit 9ff93575 (2016-05-06): Updated translation instructions, closes #230
closed 2016-05-06 by ciampix