KiCad/kicad-doc #239: GSIK: German translation is WIP

tijuca, 2015-12-24:
Hello,

I just want to prevent some double work. I've started to work on the German translation for getting_started_in_kicad. But I will need a few more days. Thanks.

ciampix, 2015-12-29:
My 2 euro cent suggestion: just commit your work, even if small. So you can ask for someone to revise it on-line reading from the multilingual compilation results...

tijuca, 2015-12-29:
Well, the translation isn't the difficult part. this is almost done. But the must boring work is to collect the respecting German menu entrys. I found there some more missed or better to translate strings in KiCad itself. After I synced the de/kicad.po to the current upstream base there was even more new strings to translate.
Also this part is mostly finished, so I will send two pull requests (one for kicad-i18n and the other for kicad-doc) shortly, hopefully yet this year. :-)

ciampix, 2015-12-29:
Ok, then may I close this PR?

tijuca, 2015-12-30:
You mean (this) Issue?
I'd like to let it open until I finished my work for the pull request so everybody else is able to see there is something going on.

ciampix, 2015-12-30:
ok

PR #259 (2016-01-04): German translation of GSIK
Please consider pulling my work for a translation of GSIK into German.
I added some additional screen shots with German locales to get relationship between the translated part and the respective screen shot.
Currently I have some trouble on chapter 4.1, point 16. I don't know why the translated string and the related screen shot isn't used.

This pull request would also close #239 if accepted.

Regards
Carsten
closed 2016-01-04 by ciampix