KiCad/kicad-doc #304: Template Section - min requirements to make a template work

scottcher, 2016-01-20:
I recently tried creating a new template and found the instructions for doing so in the documentation site to have gaps that left me wondering why my template wasn't working as expected.

I suggest adding to the required files section a project file.

When I tried creating a template, I included the info.html file and icon.png file in the meta directory but only include the file I was intending to template from in the main template folder - in my case, the PCB file.  When I used the template, what I got was a new project file with a new schema and new PCB file and a copy of my intended PCB template file.

the solution was to include a project file in the template folder but that wasn't clear from the template instructions.

ciampix, 2016-01-21:
Thanks for the heads-up. Would you point me to the document/section you are referring to?
May you suggest a text or (best) a patch?
TIA

scottcher, 2016-01-22:
Sure.  The document section I'm referring to is http://docs.kicad-pcb.org/en/kicad.html#_using_templates but not the entire section.  This section is more about how to use templates and doesn't provide much details around creating them.

I can try to write something up that might help others avoid the problem I encountered - I'd be happy to contribute.

evanshultz, 2016-06-17:
This may have been fixed. I'm able to create a new template with only the required file (info.html), with required and optional files, and with all of the previous plus project files (*.pro, *.sch, etc.). In all cases my new project has a project file, schematic file, and PCB file. My version info:
Application: kicad
Version: 4.0.2-stable release build
wxWidgets: Version 3.0.2 (debug,wchar_t,compiler with C++ ABI 1009,GCC 5.2.0,wx containers,compatible with 2.8)
Platform: Windows 8 (build 9200), 64-bit edition, 64 bit, Little endian, wxMSW
Boost version: 1.57.0
         USE_WX_GRAPHICS_CONTEXT=OFF
         USE_WX_OVERLAY=OFF
         KICAD_SCRIPTING=ON
         KICAD_SCRIPTING_MODULES=ON
         KICAD_SCRIPTING_WXPYTHON=ON
         USE_FP_LIB_TABLE=HARD_CODED_ON
         BUILD_GITHUB_PLUGIN=ON

Also on my Win10 PC, I see the templates at C:\Program Files\KiCad\share\kicad\template. Can anyone else confirm that's correct for Windows?

evanshultz, 2016-06-22:
@scottcher can you reproduce this issue using the 4.0.2 stable release? Please paste the info from Help > Copy Version Information if you can.

I'd appreciate if you could review https://github.com/KiCad/kicad-doc/blob/master/src/kicad/kicad.adoc#project-templates and share if the documentation is now easier to follow. Thanks!

scottcher, 2016-06-27:
Will give it a try. Thanks