KiCad/kicad-doc #306: Getting error during install

keruseykaryu, 2016-01-21:
I'm getting error during full KiCad build with Linux Mint Rosa:

    Scanning dependencies of target pl_editor_translate_en
    Built target pl_editor_translate_en
    Scanning dependencies of target pl_editor_html_en
    Built target pl_editor_html_en
    Scanning dependencies of target pl_editor_translate_es
    Built target pl_editor_translate_es
    Scanning dependencies of target pl_editor_html_es
    asciidoc: ERROR: pl_editor.adoc: line 640: [tabledef-default] missing closing delimiter
    make[2]: *** [src/pl_editor/CMakeFiles/pl_editor_html_es] Error 1
    make[1]: *** [src/pl_editor/CMakeFiles/pl_editor_html_es.dir/all] Error 2
    make: *** [all] Error 2

nickoe, 2016-01-21:
Same does http://ci.kicad-pcb.org/job/any-kicad-doc-head/203/console

nickoe, 2016-01-21:
I think it is working now.

closed 2016-01-21 by nickoe

ciampix, 2016-01-21:
IIRC previously it was possible to start an automatic compiling test with Jenkins simply writing a message containing "retest this please". Now it does not work anymore. Does anyone know how to re-enable it? Just tell me how to do it... I want to learn...