KiCad/kicad-doc #341: KiCad Getting Started Document

LarryJoy, 2016-04-18:
[Getting Started.pdf](https://github.com/KiCad/kicad-doc/files/222957/Getting.Started.pdf)

ciampix, 2016-04-18:
I have rewritten this feedback (thanks) to a more detailed and manageable version:

GETTING STARTED April 16, 2016
Document comments and feedback

- [ ]  Page 9, Nr 9 Note, 2 nd sentence, “You will see...these Clarify Selection menu often in KiCad, they allow working...” I believe this should be, “You will see the clarify selection menu often in KiCad, as it allows working...”

- [ ] Page 9, Nr 11, 2 nd sentence, “Replace the current value with 1k.” In keeping with SI this should be “1kΩ (Ω>Alt 2-3-4) or at least “1 k”. There is always a space between the number and the prefix symbol or unit symbol.

- [ ] Page 11, Nr 18, last sentence, “In general, it is recommendable to use...” I think this should be “In general, it is recommended to use...”

- [ ] Page 12, the figure at the top of the page. In keeping wit the standards I know, namely IEEE 315,
Clause 22.4; Class Designation letter(s) IC should be U and D should be DS.

- [ ] Page 13, figure at the top of the page. There is a better way to depict connectors electrically on a
schematic diagram, in this case MYCONN3. Reference IEC 60617.

- [ ] Page 14, figure at the top of the page:
1) There is a problem with the VCC symbol that you show “O—“. According to IEEE 315, Clause
5.1.1/IEC 60617, Clause ? The unfilled circle is the symbol for a terminal. There is no terminal.
The symbol to use is from IEEE 315A, Clause 1.1.1 One-way/IEC 60617.

[symbol image, see the pdf link] (https://github.com/KiCad/kicad-doc/files/222957/Getting.Started.pdf)

For an example of usage of this symbol see ANSI/ASME Y14.44, Fig. 3, the schematic diagram at
3A1J1-10 and 3A1AR1P1-2.

2) The newer IEEE 315A, Clause 8.5.4.2/IEC 60617 symbol for a diode or LED uses an unfilled
triangle with a line that runs between the anode and cathode.

[symbol image, see the pdf link] (https://github.com/KiCad/kicad-doc/files/222957/Getting.Started.pdf)

3) Although you use the triangle common return symbol with the word “GND”
[symbol image, see the pdf link] (https://github.com/KiCad/kicad-doc/files/222957/Getting.Started.pdf)
that is shown in IEEE 315, Clause 3.9.3.2/IEC 60617?, the better symbol to use is IEEE 315, Clause
3.9.2/IEC 60617.

[symbol image, see the pdf link] (https://github.com/KiCad/kicad-doc/files/222957/Getting.Started.pdf)

For an example of usage of this symbol see ANSI/ASME Y14.44, Fig. 3 in many places.

- [ ] Page 27, Nr 20, 2 nd sentence, “Select B.Cu in the drag down menu...” Should “drag” be “drop”?

- [ ] Page 31, 4.3, Nr 1, 2 nd sentence, “...drage down menu...” Should this be “...drop down menu...”?

- [ ] Page 31, 4.3, number 3, “...go again for the...” I think this should be “...go again to the...”

- [ ] Page 31, 4.4, Note, “Freerouter is a open...” This should be “Freerouter is an open...”

- [ ] Page 35, Chapter 6, Make schematic components in KiCad, 1 st sentence, “Sometimes a component that you want to place on your schematic is not in the KiCad libraries.” I think this would be better if stateed “...is not in a KiCad library.”

- [ ] Page 37, figure at the top of the page. It would be better to depict a connector in accordance with IEC 60617. I do not have access to IEC 60617, but I do have access to IEEE 315 that shows symbols from IEC 60617 by indicating such symbols with “ ̄ ̄ ̄IEC”. See IEEE 315A, Clause 5.3.1 for the female contact and Clause 5.3.2 for the male contact.

[symbol image, see the pdf link] (https://github.com/KiCad/kicad-doc/files/222957/Getting.Started.pdf)

Now look at IEEE 315A, Clause 5.3.4.1 Application: engaged 4-conductors (male plug – female
receptacle shown). Note that when these contacts are grouped together, as for a connector, there is a vertical line depicting mechanical linkage. I draw it as follows:

[symbol image, see the pdf link] (https://github.com/KiCad/kicad-doc/files/222957/Getting.Started.pdf)
Two or more female contacts that make up a connector.

[symbol image, see the pdf link] (https://github.com/KiCad/kicad-doc/files/222957/Getting.Started.pdf)
Two or more male contacts that make up a connector.

evanshultz, 2016-06-12:
I'm a native English speaker with experience doing technical writing. Regarding the first point, I'm looking at the December 2015 Getting Started document and I don't see this text on page 9 so I'm not sure of the context. However, it would be correct to change "these" to "the" and "they allow" to "it allows" as shown above. Capitalization of Clarify Selection could go either way, but I think it's best to capitalize and I see capitalization being used for dialog titles in the document already.

ciampix, 2016-06-12:
Ok, sorry for the delay, I was in the middle of an update from Ubuntu 14.04 to 16.04 and a change of disk of my laptop. I agree to all and I am updating now.

evanshultz, 2016-06-19:
I vote for closing this issue. All unimplemented items are related to the symbol graphics, which aren't worth changing. If a user wants IEEE-approved symbols, that can be done. But the default symbols are intelligible and need not be changed in my opinion.

The other changes suggested are great and already finished.

ciampix, 2016-06-19:
I agree, thanks to all!

closed 2016-06-19 by ciampix