KiCad/kicad-doc #343: Eeschema

LarryJoy, 2016-04-18:


LarryJoy, 2016-04-18:
[Eeschema.pdf](https://github.com/KiCad/kicad-doc/files/223005/Eeschema.pdf)

evanshultz, 2016-06-19:
"Bill of Materials" is widely understood. Same with symbols.

For other issues, @ciampix or I have done them. This was great feedback and I nominate closing.

evanshultz, 2016-06-22:
@ciampix Can this issue be closed? It appears everything has been implemented.

kinichiro, 2016-12-12:
I checked all the issues above with master branch document, and it is OK.
There are 7 unchecked still, but we won't fix these since these can't fix on documentation.
I would like to close this issue in this weekend if there is no objection :-)

nickoe, 2016-12-12:
No objection from here.

LarryJoy, 2016-12-13:
Yes, okay to close.Larry Joy

kinichiro, 2016-12-16:
Thanks for confirmation message. This is closed.

closed 2016-12-16 by kinichiro