KiCad/kicad-doc #349: Pcbnew

LarryJoy, 2016-04-27:
Pcbnew April 19, 2016
Document comments and feedback

Page 3, 22, 1st bullet point, 2nd sentence, “On Windows is in C:...” I think this should be “On Windows it is in C:...”

Page, paragraph at the bottom of the page, 1st sentence, “...and the library only require for the project...” I think this should be “...and the library only required for the project...”

Page 12, 3.6, 1st bullet point, “...deletes a footprin...” This should be “...deletes a footprint...”

Page 12, 3.6, 4th bullet point, “...display the list off...” This should be “...display the list of...”

Page 30, 4.2, 2nd bullet point, “...to the corresponding footprint (often called footprint)...” Should this be “...to the corresponding land pattern (often called footprint)...”?

Page 35, 5.1, 1st bullet point, “Between 2 and 32 copper layers...” Should this be “Between 1 and 32 copper layers...”? Or is it that Pcbnew can not do a single layer board?

Page 41, 5.5, last paragraph (last sentence), “One can also switch to an other...” This should be “One can also switch to another...”

Page 47, 6.1.2.3, 1st sentence, “Here is an example of an DXF...” This should be “Here is an example of a DXF...”

Page 49, 6.1.3, Note, “...changing a 1/8W resistance to 1/2W...” This should be “...changing a 1/8 W resistor to ½ W...”

Page 52, last bullet point, “Update all footprints of the board for reloads all footprints on board”? Think this should be “...for reloading of all footprints on the board”.

Page 57, 7.4.1, 2nd paragraph, 3rd line, “the larger footprints with with many pads.”  Should be “the larger footprints with many pads.”

Page 58, 7.4.3, 3rd paragraph, 1st line (1st sentence), “...makes it possible to re-place...” Should be “...replace...”

Page 61, 8.3, 3rd bullet point, “Users can add others Netclasses.” Should be “Users can add other Netclasses.”

Page 63, 1st sentence, “This dialog also allows to enter...and vias sizes.” Should be “This dialog also allows you to enter...and via sizes.”

Page 64, 8.3.6, 1st sentence, “...and/or vias sizes.” I think this should be “...and/or via sizes.”

Page 65, 8.5.1, 1st bullet point, “...0.35mm...” Should be “...0.35 mm...”

Page 65, 8.5.1, 2nd bullet point, “...0.8mm...” Should be “...0.8 mm...”

Page 65, 8.5.1, 3rd bullet point, “...1.91mm...” Should be “...1.91 mm...”

Page 65, 8.5.1, 4th bullet point, “...2.54mm...” Should be “...2.54 mm...”

Page 65, 8.5.1, 5th bullet point, “...2.54mm...” Should be “...2.54 mm...”

Page 65, 8.5.1 figure. Use SI properly with space between number and following SI symbol.

Page 65, 8.5.2, 1st bullet point, “...0.35mm...” Should be “...0.35 mm...”

Page 65, 8.5.2, 2nd bullet point, “...0.5mm...” Should be “...0.5 mm...”

Page 65, 8.5.2, 3rd bullet point, “...0.05 x 0.1 inches...” Should be “...0.05 x 0.1 inch...”

Page 65, 8.5.2, 4th bullet point, “...1.27mm...” Should be “...1.27 mm...”

Page 66, figure at top of page. Use SI correctly.

Page 66, 8.6, 4th line, “...positioning of footprins.” Should be “...positioning of footprints.”

Page 68, 8.8, 2nd paragraph, at the end of the sentence, “...extra tracks and vias sizes.” Should be “...extra tracks and via sizes.”

Page 68, 8.8.1, table, 2nd row, 3rd (last) sentence, “Others values are tracks widths...” Should be “Other values are track widths...”

Page 68, 8.8.1, table, 4th row, 2nd sentence, “The first value in list...” Should be “The first value in the list...”

Page 68, 8.8.1, table, 4th row, 3rd (last sentence, “Others values are vias dimensions...” Should be “Other values are via dimensions...”

Page 75, 9.5, 1st sentence, “The router behavior be configured...” Should be “The router behavior can be configured...”

Page 78, last paragraph, 2nd sentence, “...created by a left-clicking...” Should be “...created by left-clicking...”

Page 83, 10.3.3, last bullet point, 1st dash, at the end, “...to connect zones areas.” Should be “...to connect zone areas.”

Page 84, 10.3.4, “Thermal reliefs parameters”.  Should be “Thermal relief parameters”.

Page 89, 10.9.1, 1st sentence, “This is done using the button.” It appears there is a missing icon after “button”.

Page 92, 11, 1st paragraph (1st sentence), “Let us see now what are the steps necessary for the creation of the necessary files...” Huh? What? Try “Let us see now what the steps are for the creation of the necessary files...”

Page 102, The available options table, last row, box on the right, 2nd sentence, “...use specific extensions foe each file.” Should be “...use specific extensions for each file.”

Page 106, top row of icon table, right hand block, 1st line, Change “...circuit board. When...” to “...circuit board, when...”

Page 106, icon table, 4th row, right hand block, 2nd line, “...difference being that creates a library...while creates a library...” I think this should be “...difference being that it creates a library...while creating a new library creates a library...”

Page 117, 13.7, 2nd paragraph, at the end, “...100uF_16V, 100_0.5W...” Is there a problem with using SI correctly (with spaces)? This should be “...100 µF_16 V, 100 Ω_0.5 W...”

Page 122, 13.10, 1st sentence, “...to eploit the the full...” Should be “...to exploit the full...”

Page 127, 14.2, 2nd paragraph, 1st sentence, at the beginning, “To use this tools...” Should be either “To use this tool...” or “To use these tools...”

Page 129, 14.3.1, 2nd paragraph, 1st sentence, “...for the types of array.” Should be “...for the types of arrays.”

Page 129, 14.3.1, 2nd paragraph, 2nd sentence, “There are two type of array...” Should be “There are two types of arrays...”

Page 131, 14.3.2.2, 1st bullet point, 3rd (last) line. After the ending period there is an “i”. Is this just spurious and needs to be removed or is there an additional thought that needs adding?

Page 131, 14.3.2.2, 3rd bullet point, 2nd line, “...from this items number.” Should be “...from this item's number.”

Page 133, 15, 1st sentence, “Scripting allow you...” Should be “Scripting allows you...”

Page 135, 3rd bullet point, “...This method returns a list of TRACKs an SEGVIAs...” Should be “...This method returns a list of TRACKs and SEGVIAs...”

Page 136, 15.6.1, 1st sentence, “...that must be keep as it is.” Should be “...that must be kept as it is.”

Page 137, 15.7, 1st sentence, “...scripts that can be access via from the Footprint Editor.” Should be either “...scripts that can be accessed via the Footprint Editor.” or “...scripts that can be accessed from the Footprint Editor.”

ciampix, 2016-04-27:
Very good job! Thanks. I will push a patch for these points in a few minutes...

LarryJoy, 2016-04-28:
Marco,What about #342, my feedback on KiCad and #343, my feedback on Eeschema? I submitted PDF files for these but that was before I got the message about just putting the comments in the body of the message instead of an attachment. Shall I go back and submit my comments from the PDF files? actually they are OpenOffice documents that would not load and so I changed them to PDF. I also have three more sets of feedback coming when I get them typed up, they are for CvPcb, GerbView, and Pl_Editor.
Regards, Larry Joy

ciampix, 2016-04-28:
Please be patient since this is not the way we usually work (this means: it is harder do apply patches in this way). If you want to see those patches applied faster you can, in sequence of work for you and easiness for us:
1. convert the pdf into text
2. add the "tick" text to all the entries (those sequence of: "- [ ]" text)
3. break into no more than 10 entries at a time
and if you want to really easy our life, create a branch and ask for a pull: that is definitely the way to do it (and your contribution will even automatically mentioned in the git log ...)
Anyway many thanks for your contribution! It is really appreciated!

LarryJoy, 2016-05-08:
ciampix, What do you mean by "create a branch and ask for a pull"? This is all new to me and I do not know how this works. I need specific instructions.--Thanks.

ciampix, 2016-05-08:
See this: https://help.github.com/articles/creating-a-pull-request/
and this: https://help.github.com/articles/using-pull-requests/
and do not be shy, ask if you do not understand something!

I think that contributing is a great source of knowledge for new technology, so I am always undecided if I am really contributing to something or instead I owe to the community a great deal.

closed 2016-06-13 by ciampix