KiCad/kicad-doc #358: Using Travis-CI for testing pull request

kinichiro, 2016-05-08:
Hi,
This is not a issue, but just a proposal for using Travis-CI for testing PR.
Last year, PR was tested by "kicadjenkins" if build succeeds or not.
How about using Travis-CI ?

With Travis-CI,
- build os is Ubuntu 14.04, unfornunately not Fedora.
- it takes about 20 min. for each PR.
- we can't see artifacts (.html, .pdf, .epub), just build test

below is `.travis.yml` and I could verify this worked.
<pre>
language: bash
dist: trusty
sudo: required
cache: apt

before_script:
  - sudo apt-get -qq update
  - sudo apt-get install -y asciidoc pandoc gettext po4a dblatex
  - sudo apt-get install -y source-highlight
  - sudo apt-get install -y fonts-freefont-ttf fonts-vlgothic
  - sudo apt-get install -y texlive-xetex
  - sudo apt-get install -y texlive-lang-cjk
  - sudo apt-get install -y texlive-lang-cyrillic
  - sudo apt-get install -y texlive-lang-dutch
  - sudo apt-get install -y texlive-lang-english
  - sudo apt-get install -y texlive-lang-french
  - sudo apt-get install -y texlive-lang-german
  - sudo apt-get install -y texlive-lang-italian
  - sudo apt-get install -y texlive-lang-polish
  - sudo apt-get install -y texlive-lang-spanish

script: mkdir build && cd build && cmake .. && make -j4
</pre>

To work with Travis-CI,
1. sign in https://travis-ci.org/ with github kicad account
2. In Travis-CI,
   1) enable kicad-doc repository at "Accounts"
   2) At "Settings" of kicad-doc, turn ON "Build only if .travis.yml is present" and " Build pull requests"
3. put `.travis.yml` on github kicad-doc top directory.

Here is example of PR.
https://github.com/kinichiro/kicad-doc/pull/3
And here is build result by Travis-CI.
https://travis-ci.org/kinichiro/kicad-doc/builds/128656047

I'm not sure if this is required by contributors.
But at least, we can tell if submitted PR breaks the build or not before merge.

Regards,

BTW, I tried to work with "sudo: false", but it didn't work well.
Here is that version of .travis.yml, for reference.
https://github.com/kinichiro/kicad-doc/issues/2#issuecomment-217718424

ciampix, 2016-06-13:
Added your suggestion of TravisCl integration, please forgive the delay with which I replied this.
Your work is much appreciated, thanks!
If it works for a few days I will close this PR.

kinichiro, 2016-06-13:
@ciampix
Thanks a lot.
I hope this would help contributers.

ciampix, 2016-06-14:
Your work is great, it is working flawlessly. I'm closing this.

closed 2016-06-14 by ciampix