KiCad/kicad-doc #369: kicad-install.sh fails on the documentation

sarfata, 2016-06-07:
Downloaded `kicad-install.sh` from http://bazaar.launchpad.net/~kicad-product-committers/kicad/product/view/head:/scripts/kicad-install.sh

It fails to download the documentation:
````
step 5) checking out the documentation from launchpad repo...
bzr: ERROR: Not a branch: "https://code.launchpad.net/~kicad-developers/kicad/doc/".
````
What is the proper way to download the documentation now?

nickoe, 2016-06-07:
`scripts/kicad-install.sh` is really unsupported and is going to be removed.

@lachlanA seems to have decided to support it at https://github.com/lachlanA/kicad-install

closed 2016-06-07 by nickoe

sarfata, 2016-06-07:
Thanks for the link to the new version.

I am not sure if this is the right place to discuss this or open an issue for this but someone should update the main KiCad page [here](https://launchpad.net/kicad) because it links to this script (in the "Useful tools" section).